vlachoudis/bCNC #1177: Adding a new tool change policy -- TLO From Library

pomeroyb, 2019-03-01:
Currently all tool change policies assume you either have a probe to check the length of each tool using a macro that replaces M6 (policy 2-3), or assumes that every tool is exactly the same length (No Probe, policy 4).

I'd like to bring up the idea of adding a tool library that enumerates tools and allows you to set TLO offsets manually -- something like we see in [this video](https://www.youtube.com/watch?v=uNjMIIRttFE). Any system that uses repeatable collets (I personally use the TTS) would benefit from this.

sfinexer, 2019-03-02:
https://github.com/vlachoudis/bCNC/issues/1001
Yes, that's right. And if there is no sensor that can be attached to the spindle?

pomeroyb, 2019-03-02:
I'm confused... that issue doesn't seem to be at all related to this one? Tool lengths are manually measured and entered, and no sensor is necessary.

sfinexer, 2019-03-02:
@pomeroyb I meant it, the link is given to the fact that it is necessary to use a sensor.
But if there is no sensor, it is impossible to apply auto level.

pomeroyb, 2019-03-02:
Having a tool length library does not prevent you from using a probe. The probe is also a tool that's overall length is inputted into the library, and any auto level measurements can be adjusted accordingly.

In your example, if the microswitch glued to an old endmill is placed into a tool holder (Like an [ISO 40](http://sc01.alicdn.com/kf/HTB1qELvKeySBuNjy1zdq6xPxFXas/ISO-40-Tool-Holder-for-CNC-Machine.jpg) or a [Tormach TTS holder](https://tormach-inc.imgix.net/wp-content/uploads/sites/8/2018/04/27173910/32309_probe_with_holder_lrg.jpg)), the offset of that switch can be measured and is very repeatable.