vlachoudis/bCNC #1178: How can I setup bCNC the first time? Complete noob, first time CNC.

vitsensei, 2019-03-01:
Hi, so I have a XY core plotter and want to use bCNC to control it.
Now if I understand this correctly (after reading about grbl), I need to do all sort of configurations. The most important ones are, for example, [X,Y,Z] steps/mm. What else you guys think I should config, and how would I do that?

Thanks a lot.

Harvie, 2019-03-01:
You need to manualy compile GRBL to support CoreXY setup. This is bCNC support, but you are asking about GRBL, which is here: https://github.com/gnea/grbl/

closed 2019-03-01 by Harvie

vitsensei, 2019-03-01:
Thanks for the quick reply.
Is there any link showing how bCNC support corexy? I assume that's the easier way.

vitsensei, 2019-03-01:
Thanks for the quick reply.
Do you know how I can use bCNC for corexy?

Harvie, 2019-03-01:
bCNC does not care if you run CoreXY or not. It only sends the coordinates. It's GRBLs duty to get the machine to that coordinates, no matter what kind of mechanical setup you have.

Harvie, 2019-03-01:
GRBL has following line in config.h:
`// #define COREXY // Default disabled. Uncomment to enable.`

I've never tried that, perhaps it will need you to setup few other settings, but there are some hints in the file aswell. I don't know this stuff. You get proper support at grbl project.

vitsensei, 2019-03-01:
Thanks, I just asked a question on GRBL.
So, what about the [X,Y,Z] steps/mm setup? Do we also do that from the Gcode file or somewhere else in bCNC?

Harvie, 2019-03-01:
Again... bCNC does not care about mechanical setup. You have to set this up in GRBL:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#grbls-xval-settings-and-what-they-mean

But this can be done after GRBL was installed on arduino and yes, you can do it using bCNC in "Terminal" tab. But the settings are stored in Arduino/GRBL.

vitsensei, 2019-03-02:
Hi, thanks for your patience.
Things are making more sense now as I read more and more.
One thing though, after play around with bCNC, I realized that every time I tried to move on X, Y direction (using the arrow keys), the machine would move diagonally instead. And when I tried to move diagonally, it would move along X and Y axis.
Can you see any obvious mistake I'm making? Thanks.

Harvie, 2019-03-02:
> And when I tried to move diagonally, it would move along X and Y axis.

Have you compiled and flashed GRBL with CoreXY enabled?

vitsensei, 2019-03-03:
Thanks, works now haha.