vlachoudis/bCNC #1180: Install files to system paths and other issues related to downstream packaging

yarda, 2019-03-01:
I am trying to get bCNC into Fedora:
https://bugzilla.redhat.com/show_bug.cgi?id=1684603

I hit few problems which complicated downstream packaging. Reporting them because fixing them could ease packaging to other distros as well:
- it requires pyserial<=3.0.1 due to problem with windows XP, but it's unsatisfiable on recent Linux distros. Is it possible to add some conditionals there? I had to patch it out.
- some files under bCNC/locale has executable mode
- it would be nice to have PNG icon
- module files do not need Python hashbangs (I can send PR for it)
- it would be nice to install system bCNC.ini under /etc on Linux
- bCNC.xbm probably to /usr/share/bCNC
- images to /usr/share/bCNC/images
- icons to /usr/share/bCBC/icons
- locales to /usr/share/locale

Regarding the paths I don't know how to properly implement it while still being backward/Windows compatible, so I am not sending PR.

Harvie, 2019-03-01:
Does fedora have some guidelines to repackage pypi packages as rpm? Some distros have tools to automaticaly convert arbitrary python (pypi) packages to distro-specific packages.

* it requires pyserial<=3.0.1 due to problem with windows XP, but it's unsatisfiable on recent Linux distros. Is it possible to add some conditionals there? I had to patch it out.

> * some files under bCNC/locale has executable mode

Is that really a problem? I guess it's there by mistake, because some files are contributed by windows users...

> * it would be nice to have PNG icon

Tkinter does not support PNG, so only thing we can do is provide separate PNG icon just for you

> * module files do not need Python hashbangs (I can send PR for it)

I guess this is also mistake, but why this matters?

> * it would be nice to install system bCNC.ini under /etc on Linux

There is no system bCNC.ini, so obviously it can't be put in /etc. The bCNC.ini included in package is not intended to be modified in any way. It's just for internal purposes of bCNC and users should never modify it. It's mentioned in README.

> * bCNC.xbm probably to /usr/share/bCNC
> * images to /usr/share/bCNC/images
> * icons to /usr/share/bCBC/icons
> * locales to /usr/share/locale

AFAIK This would be impractical for pypi packaging... I think you can solve this by adding symlinks to your package to work around this, but that would only make your package hard to maintain, because this stuff will change in future.

onekk, 2019-03-01:
I have had no problems running bCNC dowloading the sources and the running on Debian 8.0 on an 
ARM processor, almost every day.

I've had not very big problem excluding those related to EMI and the CH340G on my Arduino that with some big gcode file seems to introduce some bug "forgetting some characters".

Regards

Carlo D.

Harvie, 2019-03-01:
@onekk why you post this here? it seems completely unrelated to this issue.

sfinexer, 2019-03-01:
I have pyserial 3.4-4.
I have never used the b-CNC installer and never installed it. I download it, unpack it, make an icon, and launch it.
I do not even know how to install it.

onekk, 2019-03-01:
Harvie, my post is only to say that I use bCNC without using Pypi and installing all the libraries using the package manager of my distribution (Debian 8 on  an ARM processor)

Why it is unrelated with the orignal post that speak about packaging bCNC on Fedora?

> it requires pyserial<=3.0.1 due to problem with windows XP, but it's unsatisfiable on recent Linux distros. Is it possible to add some conditionals there? I had to patch it out.

Altough Debian 8 is not very recent, but jurassic.

Regards

Carlo D.

QuLogic, 2019-07-18:
> > * it requires pyserial<=3.0.1 due to problem with windows XP, but it's unsatisfiable on recent Linux distros. Is it possible to add some conditionals there? I had to patch it out.
> 
> I don't know how to do this. If i put this condition to `setup.py`, it will be AFAIK evaluated on my PC when uploading to PyPI, so all users will get metadata built in Linux environment. I think that python packaging lacks such kind of flexibility, but i agree this is a PITA. I would be happy if you figure this one out!

You can specify [environment markers on markers](https://pip.pypa.io/en/latest/reference/pip_install/#requirement-specifiers) that are conditional on Python version, system platform, etc.

Harvie, 2019-07-19:
> You can specify environment markers on requirements that are conditional on Python version, system platform, etc.

Cool. This will save us some headache if it can tell if running ARM or not!

Harvie, 2019-07-19:
@QuLogic would you consider the following setup.py code correct?

```
	install_requires = [
		"pyserial ; sys_platform != 'win32'",	#Windows XP can't handle pyserial newer than 3.0.1
		"pyserial<=3.0.1 ; sys_platform == 'win32'",
		'numpy>=1.12',
		'Pillow>=4.0',
		'opencv-python>=2.4 ; "arm" not in platform_machine',	#Note there are no PyPI OpenCV packages for ARM (Raspberry PI, Orange PI, etc...)
	],

```

QuLogic, 2019-07-25:
It seems reasonable to me.