vlachoudis/bCNC #1183: Help to call other plugin

MARIOBASZ, 2019-03-07:
How do I call the profile operation from a plugin?
The writing
app.gcode.profile (1,1)
but it is wrong
I would also like to have the option to be able to call a trochoid_path from trochoidal, and to generate the result on this route, it can not be done as.
How do I make the generated name be xxx and the name of the original block? (example troch3d_originalnameofblock)

Harvie, 2019-03-07:
You have to first convert g-code to path. Eg.:

```python
                for bid in app.editor.getSelectedBlocks():
                        if len(app.gcode.toPath(bid)) < 1: continue

                        opath = app.gcode.toPath(bid)[0]
```

closed 2019-03-07 by Harvie

MARIOBASZ, 2019-03-07:
I did not understand, how do I place the profile call?