vlachoudis/bCNC #1185: I need a little help to add cut operation to Trochoid_path

MARIOBASZ, 2019-03-08:
I am completing another trochoidal version: trocoidal3d, with adaptive diameters in the corners
Basically the operation is to follow the selected route, be it flat or in 3d.
If I want to cut a figure, today I must perform the following operations:
1-I select the block
[image]

2- Generate the route (in this case inside adaptive)

[image]
I must take the precaution of maintaining the same selection of cutting diameter and bit used

3- Cutting operation for the depth of cut and every how much (suppose you want to cut 10mm of aluminum in 5mm passes: cut -10 5)
4-Correr plugin trocoidal3d
[image]
[image]

[image]
 
It descends helically, and the descent per turn depends on the feed and zfeed speeds

The ideal is to do everything with just one click. I need to add the values of Z objective (cut) in trochoid_path, to then call trochoid_path from Trochoid 3d (maybe not elegantly, but at least I can do it, although I am grateful for help).

MARIOBASZ, 2019-03-09:
I have made progress, you can add cut in the selected steps.
I'm blocked to add tab as islands and it works:
What does work:
I have modified trochprofile in cnc file:
after
del blocks [:]
blocks.extend (newblocks)
  I add
self.cut (reversed (blocks), -2, 1,0,900,120.0, 0,0,0,1,0,0,0,0,0)
  and I get to generate the cut operation correctly.

The problem:
I get to add tab / islands, but in that case then cut does not run
If I add create tab before cut:
self.createTabs (reversed (blocks), 1, 0, 1, 0, -1.2,1)
self.cut (reversed (blocks), -2, 1,0,900,120.0, 0,0,0,1,0,0,0,0,0)

there is no result except the route before cut, and the created block of tab
(I suppose the error is due to not being able to select the generated path file again). Any help? I think with knowledge of python it should be easy

Harvie, 2019-03-10:
This does not look very square:

[image]

MARIOBASZ, 2019-03-10:
@Harvie, if you want it rectified, I do not remember if you can make the adaptive trochoids are also rectified, I think not (maybe you can think of how to improve it). I do not know if the commercial programs generate for this type of cut only pure trochoids or on the contrary rectified as you have done (and I added to my plugin). I think it is faster to generate trochoidal cutting and then pass rectification if necessary: in my machine I have bounded advance between trochoids of 0.1 or 0.05mm to be able to mill aluminum in one pass, with which the roughness can be negligible. Now that I think about it, the distance between trochoids in the adaptive sector could be further diminished to improve the termination! This image is very enlarged (ae = 0.1 I think it is)

MARIOBASZ, 2019-03-10:
The valleys are too pronounced, since they are arcs of circles that progressively diminish the radius: first, I thought: Have you made an error? but I think that it is due (or in the viewer, or in reality) to the "square of the circle" that the viewer performs (or the real movement)

MARIOBASZ, 2019-03-10:
Anyway I think it is a good pisibilidad to increase the number of trochoids in adaptive corners: at present the speed of advancement decreases as the radius decreases (intuitively razobe that I protect the tool by greater contact arc), but it is an option to diminish (4 times?) The distance between trochoids in adaptive. I'll think about it.

closed 2019-03-11 by MARIOBASZ