vlachoudis/bCNC #1186: When will bCNC appear in distributions?

sfinexer, 2019-03-10:
When will bCNC appear in distributions?
https://www.debian.org/distrib/packages

It is not possible to install bCNC on a computer using standard methods.

Harvie, 2019-03-10:
> When will bCNC appear in distributions?
> https://www.debian.org/distrib/packages

Package maintainers of these distributions have to decide to include bCNC themselves.

There is some work being done to add bCNC packages to Fedora: #1180 
Also there are packages for ArchLinux:
https://aur.archlinux.org/packages/bcnc/
https://aur.archlinux.org/packages/bcnc-git/

> It is not possible to install bCNC on a computer using standard methods.

This is just not true. You can install latest bCNC using pip on any OS running python (see docs).
It's even better than using distribution specific package manager for following reasons:

1.) You get latest version immediately (no need to wait for distro maintainers to package new version)
2.) It works on all OS including Windows and Mac that don't ship with package manager by default
3.) You can easily install older versions od bCNC if needed

closed 2019-03-10 by Harvie

sfinexer, 2019-03-10:
This is true for advanced users.
But the user will not always install something on his own (especially when he first encounters a CNC machine)

Harvie, 2019-03-10:
> This is true for advanced users.

I don't know what are you talking about.
I don't think it's that different: `apt install bcnc` vs `pip2 install bCNC`

sfinexer, 2019-03-10:
>You get the latest version immediately (no need to wait for distro maintainers to package new version)

I think this will be done automatically. 

I gave a link to search for packages.
bcnc is not in the debian package list.
When a person begins to learn CNC he goes the easiest way. And in the case of Debian, he cannot install it using the package manager.
https://packages.debian.org/bcnc

Harvie, 2019-03-10:
I am not against including the bCNC in the Debian. But i don't use Debian on my laptop. I think that Debian is not good distro for desktops, so i only use it on servers. Nor i am Debian maintainer.

If you are Debian user, then you can try and become Debian mainainer. Once you are maintainer of Debian, you can include bCNC.

You can also send me pullrequest which will add files needed to make Debian package. eg. in "debian" subdirectory in root of this repo.

sfinexer, 2019-03-10:
You're right. Therefore, I have never used Debian stable releases. Only test branches.
But I think there is a way out by making your own separate repository as linuxcnc, virtualbox, skype, wine ....

In this case, we will not depend on the main repository, but will have our own.
But I never did it (((.
The advantage of your own repository is that the update will be automatic.

sfinexer, 2019-03-10:
For the sake of interest, I tried to install bCNC for the first time as described.
https://github.com/vlachoudis/bCNC#installation-using-pip--reccomended
I chose a team at random, that's what happened:

> sfinexer@sfinexer:/media/files/выжигание/bCNC$ pip2 install --upgrade git+https://github.com/vlachoudis/bCNC
Collecting git+https://github.com/vlachoudis/bCNC
  Cloning https://github.com/vlachoudis/bCNC to /tmp/pip-req-build-2BQLNY
Requirement already satisfied, skipping upgrade: Pillow>=4.0 in /usr/lib/python2.7/dist-packages (from bCNC==0.9.14.53) (5.4.1)
Requirement already satisfied, skipping upgrade: numpy>=1.12 in /usr/lib/python2.7/dist-packages (from bCNC==0.9.14.53) (1.16.1)
Collecting pyserial<=3.0.1 (from bCNC==0.9.14.53)
  Downloading https://files.pythonhosted.org/packages/42/68/41bab8a5e3e50705f25358f043181242fa197f313aa0476908d9852c8dc4/pyserial-3.0.1.tar.gz (134kB)
    100% |████████████████████████████████| 143kB 1.4MB/s 
Building wheels for collected packages: bCNC, pyserial
  Running setup.py bdist_wheel for bCNC ... done
  Stored in directory: /tmp/pip-ephem-wheel-cache-zOT1vi/wheels/9e/e3/87/8dd7776083a4daaf9c62b3c5ca05715830e522a30ce078990a
  Running setup.py bdist_wheel for pyserial ... done
  Stored in directory: /home/sfinexer/.cache/pip/wheels/70/9f/37/1ac95f9cd087c10043bbe7847ba263e9331bc186fc64a7f014
Successfully built bCNC pyserial
Installing collected packages: pyserial, bCNC
  The script bCNC is installed in '/home/sfinexer/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed bCNC-0.9.14.53 pyserial-3.0.1

Nothing has happened. Or I do not understand where b-cnc was established ....

Harvie, 2019-03-10:
@sfinexer oh dear. you have long term troubles reading documentation. it's right on the front page of bCNC project here on github:

[image]

sfinexer, 2019-03-11:
@Harvie  Yes, it starts like this ......... I am very surprised.
I always just copied git.
But then the question is, where is bcnc now?

ThierryM, 2019-03-11:
Hi,
Now many applications are distributed under "universal" packages like flatpack, snap or appimage (QGis, Openboard, ...). Maybe a solution ? More details here : https://askubuntu.com/questions/866511/what-are-the-differences-between-snaps-appimage-flatpak-and-others
Regards,

Thierry

Harvie, 2019-03-11:
> But then the question is, where is bcnc now?

It is installed along with other python modules, just like pyserial or numpy. Where it exactly goes, dependes on your system.

> Now many applications are distributed under "universal" packages like flatpack, snap or appimage

I know about this. I am thinking about AppImage, but it's unlikely to happen soon. First i want to get automatic builds of .exe files for windows. AppImage might be next step, but i don't think that it's really necessary, since pypi/pip seems to work effortlesly on all platforms.

sfinexer, 2019-03-11:
hmm ...
And what if we open a certain "progress bar", say "to pay for your own server", or something like that?
But here it is necessary to think whether there are some donations at the current moment.
In general, there may be no need for automatic assembly.
But I think that the development cycle of the project is necessary, for example, every half a year a new version. This algorithm is used by many programs (for example, Windows ... which releases a new version every half year)

It is necessary to release a stable version as soon as possible, errors can always be corrected.
@Harvie , you say that debian releases releases for a long time, but does bCNC release them often? Thus it turns out that you contradict yourself.

Harvie, 2019-03-11:
>  And what if we open a certain "progress bar", say "to pay for your own server", or something like that?

I don't need money for server. I manage tens to hundreds of servers as my day job, so it's no problem to host the project.

On the other hand, if you need some new feature and want to pay for it, there are websites, where you can place so called "bounty" money on github projects and developer who manages to implement the feature gets the money. Anyone can open such request. One of such websites is https://www.bountysource.com/

Feel free to use that. If you manage to raise money and pay developers using bountysource, i will be happy to merge new features to bCNC!

> In general, there may be no need for automatic assembly.

There is such need. Because people complain about some problems and i need to provide them with nightly builds, so they can check immediately if the problem was fixed. Even long before stable release.

> stable version as soon as possible, errors can always be corrected.

But you can't call it stable then :-)

>  you say that debian releases releases for a long time, but does bCNC release them often? 

don't worry we are moving fast for stable release :-) when there was the last release, i wasn't even developer of bCNC, so i was not able to do the release.

Here you can see what is left to be done until next release:
https://github.com/vlachoudis/bCNC/milestones

Here you can find discussion about releases in general:
https://github.com/vlachoudis/bCNC/labels/release%20engineering

yarda, 2019-07-18:
Regarding Fedora, I opened:
https://bugzilla.redhat.com/show_bug.cgi?id=1684603

But currently I am not allowed to package it due to lack of python3 support, so waiting for python3 version :) For python2 (because it's phasing out in Fedora) it would require FESCO exception which I didn't obtain.

Harvie, 2019-07-19:
@yarda I've tried several times to port bCNC to python3 and failed. Do you know how to do this? It seems there are some rather nasty issues with unicode, which i don't really understand... We've done some preparation work that was needed for migration to python3, but there are still some issues left. I think it mainly boils down to this:

python3 uses unicode encoding, GRBL (over serial) does not use unicode encoding, TK/Tkinter also uses something, but i am not sure what. I haven't been able to properly convert between these.

We'll be happy if you manage to solve this.

yarda, 2019-07-19:
I will take a look. We ported several projects to python 3 in the past, but this one can be really tricky :)

Harvie, 2019-07-19:
> We ported several projects to python 3 in the past, but this one can be really tricky :)

Actually that sounds quite good. To be honest i've started using python when i've recently came to (at the time) semi-abandoned bCNC project and basicaly learned python from bCNC sources, i have no other experience with python at all and never succesfully ported anything to python 3 (nor i've written code in python3), so your chances are WAAAY higher :-)