vlachoudis/bCNC #1190: TypeError: comports() takes no arguments (1 given)

smooker, 2019-03-12:
File "/mnt/vg3.2/home/smooker/src/bCNC/bCNC/FilePage.py", line 333, in comportRefresh
    for i in comports(include_links=True):
TypeError: comports() takes no arguments (1 given)

diff --git a/bCNC/FilePage.py b/bCNC/FilePage.py
index 3b0a241..3c1a8d5 100644
--- a/bCNC/FilePage.py
+++ b/bCNC/FilePage.py
@@ -330,7 +330,7 @@ class SerialFrame(CNCRibbon.PageLabelFrame):
        def comportRefresh(self, dbg=False):
                #Detect devices
                hwgrep = []
-               for i in comports(include_links=True):
+               for i in comports():
                        if dbg:
                                #Print list to console if requested
                                comport = ''
@@ -340,11 +340,11 @@ class SerialFrame(CNCRibbon.PageLabelFrame):
                                hwgrep += ["hwgrep://"+hw+"\t"+i[1]]
 
                #Populate combobox
-               devices = sorted([x[0]+"\t"+x[1] for x in comports(include_links=True)])
+               devices = sorted([x[0]+"\t"+x[1] for x in comports()])
                devices += ['']
                devices += sorted(set(hwgrep))
                devices += ['']
-               devices += sorted(["spy://"+x[0]+"?raw"+"\t(Debug) "+x[1] for x in comports(include_links=True)])
+               devices += sorted(["spy://"+x[0]+"?raw"+"\t(Debug) "+x[1] for x in comports()])
                devices += ['', 'socket://localhost:23', 'rfc2217://localhost:2217']
 
                #Clean neighbour duplicates

Harvie, 2019-03-12:
What version of pyserial do you have?

smooker, 2019-03-12:
smooker@sh21 ~/src/bCNC $ ls -la ~/.local/lib64/python2.7/site-packages/pyserial-3.0.1-py2.7.egg-info/

Harvie, 2019-03-12:
Can you please try latest git version?

smooker, 2019-03-13:
Nope. Here is the output...
```
smooker@sh21 ~/src $ git clone https://github.com/vlachoudis/bCNC.git bcnc.fixed
Клониране и създаване на хранилище в „bcnc.fixed“…
remote: Enumerating objects: 72, done.
remote: Counting objects: 100% (72/72), done.
remote: Compressing objects: 100% (44/44), done.
remote: Total 9111 (delta 40), reused 52 (delta 28), pack-reused 9039
Получаване на обекти: 100% (9111/9111), 42.14 MiB | 3.47 MiB/s, действието завърши.
Откриване на съответните разлики: 100% (5737/5737), действието завърши.
smooker@sh21 ~/src $ cd bcnc.fixed
smooker@sh21 ~/src/bcnc.fixed $ python2.7 ./bCNC.py 
This is currently broken. Use instead: python -m bCNC
bCNC runpy loader: /mnt/vg3.2/home/smooker/src/bcnc.fixed/bCNC/__main__.py
new-config Utils <ConfigParser.ConfigParser instance at 0x7fb3beba0248>
self.canvas.winfo_id(): 46137431
Traceback (most recent call last):
  File "./bCNC.py", line 20, in <module>
    runpy.run_path(bcncpath, run_name='__main__')
  File "/usr/lib64/python2.7/runpy.py", line 252, in run_path
    return _run_module_code(code, init_globals, run_name, path_name)
  File "/usr/lib64/python2.7/runpy.py", line 82, in _run_module_code
    mod_name, mod_fname, mod_loader, pkg_name)
  File "/usr/lib64/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/mnt/vg3.2/home/smooker/src/bcnc.fixed/bCNC/__main__.py", line 2675, in <module>
    main()
  File "/mnt/vg3.2/home/smooker/src/bcnc.fixed/bCNC/__main__.py", line 2643, in main
    application = Application(tk)
  File "/mnt/vg3.2/home/smooker/src/bcnc.fixed/bCNC/__main__.py", line 196, in __init__
    page = cls(self.ribbon, self)
  File "/mnt/vg3.2/home/smooker/src/bcnc.fixed/bCNC/CNCRibbon.py", line 115, in __init__
    self.register()
  File "/mnt/vg3.2/home/smooker/src/bcnc.fixed/bCNC/FilePage.py", line 384, in register
    (SerialFrame,))
  File "/mnt/vg3.2/home/smooker/src/bcnc.fixed/bCNC/CNCRibbon.py", line 134, in _register
    w = f(self.master._pageFrame, self.app)
  File "/mnt/vg3.2/home/smooker/src/bcnc.fixed/bCNC/FilePage.py", line 253, in __init__
    self.comportRefresh()
  File "/mnt/vg3.2/home/smooker/src/bcnc.fixed/bCNC/FilePage.py", line 333, in comportRefresh
    for i in comports(include_links=True):
TypeError: comports() takes no arguments (1 given)
```

smooker, 2019-03-13:
```
smooker@sh21 ~/src/bcnc.fixed $ git apply -v ../bCNC.patch 
Checking patch bCNC/FilePage.py...
Applied patch bCNC/FilePage.py cleanly.

smooker@sh21 ~/src/bcnc.fixed $ !py
python2.7 ./bCNC.py  -d
This is currently broken. Use instead: python -m bCNC
bCNC runpy loader: /mnt/vg3.2/home/smooker/src/bcnc.fixed/bCNC/__main__.py
new-config Utils <ConfigParser.ConfigParser instance at 0x7f8c53558dd0>
self.canvas.winfo_id(): 46137431

smooker@sh21 ~/src/bcnc.fixed $ cat ../bCNC.patch 
diff --git a/bCNC/FilePage.py b/bCNC/FilePage.py
index 3b0a241..4c5cc09 100644
--- a/bCNC/FilePage.py
+++ b/bCNC/FilePage.py
@@ -330,21 +330,21 @@ class SerialFrame(CNCRibbon.PageLabelFrame):
 	def comportRefresh(self, dbg=False):
 		#Detect devices
 		hwgrep = []
-		for i in comports(include_links=True):
+		for i in Utils.comports(include_links=True):
 			if dbg:
 				#Print list to console if requested
 				comport = ''
-				for j in i: comport+=j+"\t"
+				for j in i: comport += str(j)+"\t"
 				print(comport)
-			for hw in i[2].split(' '):
-				hwgrep += ["hwgrep://"+hw+"\t"+i[1]]
+			for hw in str(i[2]).split(' '):
+				hwgrep += ["hwgrep://"+hw+"\t"+str(i[1])]
 
 		#Populate combobox
-		devices = sorted([x[0]+"\t"+x[1] for x in comports(include_links=True)])
+		devices = sorted([str(x[0])+"\t"+str(x[1]) for x in Utils.comports(include_links=True)])
 		devices += ['']
 		devices += sorted(set(hwgrep))
 		devices += ['']
-		devices += sorted(["spy://"+x[0]+"?raw"+"\t(Debug) "+x[1] for x in comports(include_links=True)])
+		devices += sorted(["spy://"+str(x[0])+"?raw"+"\t(Debug) "+str(x[1]) for x in Utils.comports(include_links=True)])
 		devices += ['', 'socket://localhost:23', 'rfc2217://localhost:2217']
 
 		#Clean neighbour duplicates
smooker@sh21 ~/src/bcnc.fixed $
```

Harvie, 2019-03-13:
```
-		for i in comports(include_links=True):
+		for i in Utils.comports(include_links=True):
```

Utils.comport is only fallback. It does not make any sense to use it by default... I know how to fix, moment pls...

Harvie, 2019-03-13:
Do you see this message? `This is currently broken. Use instead: python -m bCNC`
It means, that you are loading bCNC wrong way.

Harvie, 2019-03-13:
Should be OK now. Please try.

closed 2019-03-13 by Harvie

smooker, 2019-03-14:
Excellent ! :)
Best wishes.

commit a440840f (2019-03-13): Compatibility with old pyserial version. Fixes #1190