vlachoudis/bCNC #1191: Further development of B-cnc.

sfinexer, 2019-03-13:
I propose to think about the version of bCNC 2.0
I think we need to restructure the program.
I have a few suggestions that I suppose may be useful in further development.
1) Select the main functions in a certain core of the program.
2) These functions should be singled out in a separate code, for example, C ++, for speed of execution, and concluded in a separate library.
3) Opening a large G-code file should use multi-threading; to do this, it is necessary to abandon the ADDITION option in the opening code of the cnc file. Instead, I suppose to run two (or more) independent threads that will add their part of the file to the cnc command structure.
4) You must enter the hot keys, as well as the option to configure them.

Here is my vision for the development of B-cnc.

I unfortunately not so strong in Python ... I do not understand the ideology of this programming language and its obstraction.

Harvie, 2019-03-13:
> Here is my vision for the development of B-cnc.

I also have big vision of bCNC (btw very similar to your vision). That's not the problem. Problem is who will write the code.

> To allocate the main functions in a certain kernel of the program.

Yes. But it's gonna be lot of work.

> Opening a large G-code file should use multi-threading;

That's just not how it works. Multithreading will not help with speed at all. Also i have 4 CPUs, but bCNC is so slow, that 4x speed up is not enough. Loading files is not slow. Processing and displaying is slow. Also i think that some stuff like tkinter canvas can't be accessed from multiple threads at once, because it's not thread safe and will result to crash.

There's no need to use multithreading, if you process data using python libraries that are optimized in C or ASM and can do multithreading automaticaly.

But i agree that there are some cases where we should use multithreading. Eg.:

 * GUI should have it's own thread so it never stops responding to button clicks
 * OpenGL visualiser should probably also have it's own thread so it does not block UI when rotating big objects
 * CAM Processing should be done in separate thread. Maybe each g-code block can be processed separately.

> These functions should be singled out in a separate code, for example, C ++, for speed of execution, and concluded in a separate library.

Yes and no. There are ready made python libraries written in C, which we can use to speed everything up without need to write that.

Namely
  * OpenCAMlib (CNC related geometry)
  * Shapely, Sympy.geometry (Generic geometry)
  * Numpy, Scipy (Processing large amount of mathematic calculations)
  * Vispy, Glumpy, Mayavi, PyQT graph, GLVis, ModernGL (GPU accelerated visualisation)
  * GTK3 (Portable GUI with Wysiwyg editor and code independent UI definition)

These are some of the ready made python libraries that can be used to dramaticaly speed up bCNC. This is how modern python software works:

 - You load file to "numpy array", which is python object, but all the data are stored in C data structures
 - You manipulate the data using numpy and scipy procedures written in C
 - You use one of the OpenGL based visualisers which can directly read and 3D display data from numpy array (again without data actualy being processed by python code, everything is in C)
 - If you want to eg. do 3D rotation of view of tens of thousands of points, you can still do it using numpy, which is very fast compared to python.
 - Then you are going to save the g-code file, so you use python code to read numpy array.

This means you only use python code in some specific cases like parsing, generating and streaming the g-code. While everything else is done in C. Also these libraries are highly optimized and some of them are even able to automaticaly process the data on multiple CPUs.

Problem is that to implement all of this we can just start completely new project. Because to do this right, we will need to rewrite 100% of bCNC. Also the OpenGL visualisers don't have support for Tkinter, so it's quite dependent on GTK3 port. So i think i will start to develop such project in parallel.

>  I do not understand the ideology of this programming language and its abstraction.

Ideology of Python is that it's easy to use by absolute beginners. As result it's unbeleivably slow and completely useless unless you write everything in C and just call it from python. Nice for easy tasks, unusable for anything big.

sfinexer, 2019-03-13:
I understand that the work is great and sometimes very difficult.
I myself have some pretty old programs of my own, and they are on delphi. Because of their complexity, there is no reason to rewrite them in the same C++.
My programs are modeling thermal and hydraulic objects in different modes. For example, the active zone of a nuclear reactor. There are my programs in C++, and purely for calculation without UI (GUI/UI are separate third-party programs)

But at the beginning, as you rightly said, it is necessary to accelerate the display speed.

Harvie, 2019-03-13:
Anyway. I don't see the point of this github issue, unless you have time to contribute changes that you propose...

closed 2019-03-14 by sfinexer

MARIOBASZ, 2019-03-14:
I read them with attention.
I think they are doing a great job pushing improvements to this soft, and I'm glad to collaborate in what I can.
It would be great to be able to improve the theme of speed, both to run plugins and for visualizations and others.
Meanwhile, I use the utilities with many benefits and I can often accelerate with them processes:
For a project of an idea that came to me (wish me luck, it is an idea of a different form of transmission of precision movement), I used Arc Fit to go from a block of 88,000 lines to 795 lines with negligible error in the silhouette, and allows me to generate faster union operations, difference, profile.
Hugs

Harvie, 2019-03-14:
> I used Arc Fit to go from a block of 88,000 lines to 795 lines with negligible error in the silhouette

Yes! I've written arcfit to optimize the code just like that. Problem is that it works only in 2D. But i think the algorithm might be extended to work also with Z.

MARIOBASZ, 2019-03-14:
I think it would be good. Another good option might be to rotate on a xz or yz plane, I have not needed it yet, although I think that rotating 90 degrees can be done by changing the selected plane

pabloinigoblasco, 2019-03-17:
Hello. In the context of double side pcb making, I think it would be nice to be able to rotate+translate the XY plane from a list of a points selected manually and another set of known points.

This would avoid the need to create mirror holes for double side printing and also recovering from crashes, and moving the pcb sheet from its initial position at the middle of the process.

For example I imagine this situation:
- After turning the pcb over, the translation and rotation of the plane changed. 
- Since I did not created the mirror holes and anchors position changed introduced some error and now the backside cannot be milled.
- However, I able manually to reach to a set of known points (controlling the position of the endmill and checking it visually with relatively high precision). Then I store the current coordinates of those known points into a list (push action). (this is the set of points A)
- Now I specify manually the theoretical coordinates of each of those points (it could even be semi-automated, for instance selecting those points as drills on the current gcode-plot) (this is the set of points B)
- Then I compute the transformation matrix that convert the set A to set B. 
- Finally I apply to all the Gcode XY coordinates this transform. 
- Now the backside pcb is aligned and I can mill the backside

We could use opencv. There  are some functions primitives that could help us to make this. I am thinking on findRigidTransform and cvWarpAffine

Harvie, 2019-03-17:
@pabloinigoblasco 

1.) This is off topic, as this issue is more about status quo of the whole project rather than about single feature you would like to add
2.) This is already possible to do with bCNC. This feature is called "orient" and you can find it under probe tab. BTW it uses some simple math rather than opencv. AFAIK opencv works with bitmaps, while g-code is vector format.

MARIOBASZ, 2019-03-18:
Pablo, if you could make improvements, it would be geandioso!
in relation to what you say, to be able to use artificial vision to locate edges / limits, perform visual probing to find the center, if you can do it, it would be very useful (maybe you fantasize)

onekk, 2019-03-20:
To be costructive and propositive.

I think, that the future of bCNC, is not to be a CAM software, for at least two reason:
- There are more robust CAM solution even free like FreeCAD, or with a small cost like CamBam wich run also on Linux.

- Developing it in Pyhton is very difficult. (I've tried in the past to achieve a similar goal, but it has proved to be a very difficult task)

-- OpenCAMlib is not very well supported by the author (I have tried to use it in my past project, but the author was very busy with his "for money job" and the collaboration was very slow) as a side note it is used in FreeCAD but if i rember well some of the code was modified by them.

I think that focusing on some little things are more "productive":

1) Some rethinking of the interface (many use bCNC on a Raspberry Pi so maybe some tweaking is needed to make it appear good on small screens 1024x768 or even 800x600 was a good goal)

2) having the ability to permit the user buttons to launch an external program (eventually passing some parameters stored by bCNC like a bunch of positions or maybe a probed grid) and then load the result in the editor to launch it using the usual (|> button).

3) make it behave like the old CNC controller when you can memorize A and B and cut a line, or a pocket directly from bCNC, having the ability to modify on the fly the X,Y and Z position of the simple generated gcode (a line or a rectangle and a facing operation are enough, maybe some circles):

mem A, memB for a line are the starting and end point, 

memA and memB for a rectangle are the boundaries (minX,minY)  and (maxX,maxY)

memA is the center and memB is the radius of a circle (maybe with the ability to cut a "circular pocket")

This is not a bigger task to achieve if some internals of bCNC are better explained or documented.

This will make more easy to mantain the base bCNC and eventually develop a "plugin protocol" to interface with some "external plugins".

To clarify my position, I've written some plugins for CamBam and I've written also some code for FreeCAD.

Today I've spend some time to resurface my spoilboard, and if bCNC would be able to do the square pocketing operation explained above it will be a matter of minute job:
1) move the tool to Xmin, Ymin, press memA
2) move the tool to Xmax, Ymax, press memB
3) press procket button
4) enter the target depth and maybe step over and step down values
5) push the big |> and the work was done.

Best Regards

Carlo D.

Harvie, 2019-03-20:
> make it behave like the old CNC controller when you can memorize A and B and cut a line, or a pocket directly from bCNC, having the ability to modify on the fly the X,Y and Z position of the simple generated gcode (a line or a rectangle and a facing operation are enough, maybe some circles):

I don't know why people keep asking for this. It's already possible #929

pabloinigoblasco, 2019-03-20:
> @pabloinigoblasco
> 
> 1.) This is off topic, as this issue is more about status quo of the whole project rather than about single feature you would like to add
> 2.) This is already possible to do with bCNC. This feature is called "orient" and you can find it under probe tab. BTW it uses some simple math rather than opencv. AFAIK opencv works with bitmaps, while g-code is vector format. There is some documentaion in wiki, but it's probably bit lacking. You can help to improve it.

1) I see
2) Nice it already exists. 
Well the opencv funcionality I mentioned are just some algebra routines that can be applied to a image or any set of points (ie a list of points - vectorial)

onekk, 2019-03-23:
> > having the ability to permit the user buttons to launch an external program
> 
> Agree this might be usefull. I guess it can be done as plugin.

I think of this directly in the "user button" maybe as a macro, rather than in a plugin.

Speaking of interface, the toolkit maybe has to be rather lightweight than GTK3, in effect many GTK developer still use GTK2, even if it is not developed anymore.

Regards

Carlo D.

Harvie, 2019-03-23:
> in effect many GTK developer still use GTK2, even if it is not developed anymore.

Yes. But i am not sure if it makes sense to start new project using unsupported GTK version. But i get your point... GTK2 UI has the classic minimalistic look, while GTK3 is using bigger widgets, since it's heavily optimized for touchscreens...