vlachoudis/bCNC #1193: Auto Level - applying incorrect values

Marcustuck, 2019-03-15:
Hi,
I am quite new to this software so sorry if it’s me messing up!
I had Auto level working last week, but this week it has stopped applying the corrections correctly for me. 
I had an automatic update to windows, update Win 10 Pro V1803 KB4489868 but I’m not sure if that would cause an issue.
I have Python 2.7.15 (64 bit on win32) – should I upgrade?
I have re-installed the bCNC software using “pip2 install --upgrade bCNC” to version 0.9.14-dev but still have the issue.
Hopefully you can see in the image below the auto-level data varies between approximately 0.3 and 0.05 mm but you can see in the first block of G-code a value about 48 is being applied to the G0 movement which is sending my Z axis to the limit switch!  (I only have 40 mm of Z)
Even when I get past this first section with the high G0, I have also seen that the correction does not seem to be applied correctly across the whole board anymore.
I have tried to find more documentation on the auto level function, but all the documents and videos have older versions of software, I’m sure it is probably a setting or something I can correct but I can’t work it out.
Any help would be much appreciated.
(Note: I normally scan with a 2 x 2 mm grid when producing the PCB, I’m using this larger grid to make the scan quicker while I try to solve the issue)
[image]

Harvie, 2019-03-15:
Hi, can you try to autolevel greater area? Lets say to add 10-20mm to all 4 sides of autoleveled rectangle... Just to make sure there's some margin of autolevel data around the g-code.

BTW i can't see anything wrong with the picture... Is there some bug to be seen in the screenshot?

Marcustuck, 2019-03-15:
Hi Harvie,
Ok, I'll try auto level on the PCB edge cut size so there is a bigger margin from the PCB tracks.

In the picture if you look at the G-code about half way down the Z should be 2.54 mm corrected by the auto level, so in the region of 2.59 to 2.84, but the software has made it 50.8137mm!!!!

Marcustuck, 2019-03-15:
Ok, so I increased the scan area but dosen't change it.  Here is a screen shot before applying auto-level so you can see the unmodified G-code...
[image]

Marcustuck, 2019-03-15:
and with the auto level applied and the big Z values added....
[image]

Harvie, 2019-03-15:
> and with the auto level applied and the big Z values added....

BTW do you know that you should not apply the autolevel? It is applied automaticaly when bCNC sends the g-code to your machine. So if you apply it manualy and autolevel grid is still loaded in bCNC while sending, it will get autoleveled twice! Manual autolevel (the button that actualy modifies g-code in editor) is meant only for exporting the g-code.

Marcustuck, 2019-03-15:
Hi Harvie,

Ok that sounds like it could be my issue then.  My wife has just made me pack up as we are going out to dinner.  If you can bear with me I will check out your suggestion tomorrow and post up the result and forward zipped examples if your suggestion hasn't worked for me.

Thanks again, 

Marcus

Harvie, 2019-03-15:
Sure. BTW while you should not apply the autolevel manualy, i think that it's not the real problem. Because the g-code in editor is autoleveled only once. The second autolevel is only sent to GRBL and not displayed at all. So that's not the problem.

I think that problem might be in the fact that your g-code block does not contain any Z coordinates. Try adding Z to the first line, or even add some g0 entry (go to safe Z, rapid to XY, feed to material Z). That might help.

Marcustuck, 2019-03-16:
Hi Harvie,

Thanks very much for your help, it was me not understanding the software correctly and effectivly adding the autolevel data twice!  So I guess this snag can be closed, thanks.

For others reading this thread, here are my instructions for Autolevel:-
1.	Load file for area to be scanned
2.	Move probe to origin you want to use and set XYZ to zero
3.	Go to Probe tab
4.	Carry out a probe in that location and accurately set Z to zero
5.	Go to Autolevel tab
6.	Clear probe data
7.	Set margin and adjust Autolevel XYZ and steps values as required
8.	Zero – so that the probe in the origin location is the bases for Autolevel calculations
9.	Scan
10.	Once scan complete just run the G-code (don’t use the Autolevel button to modify the G-code)

You can see the autolevel corrected Z values (in terminal) in the screen shot below.  In my G-code the Z is set to 0.05 mm and you can see that bCNC has automatically adjusted this value as the Autolevel data is showing my board is higher at that milling point - Great system :-)

I also noticed that the autolevel was reading the origin as about 0.013 mm higher than my probe for that loaction, Z feed rate set the same for the autolevel scan and probe, so I will add a small allowance to my Z to allow for this small difference.

I have also attached my G-code and probe files if you want to see why when you manually added Autolevel the Z jumps by 48 mm during transit movements, I suspect there is a default safe moving Z somewhere that is being applied. 

[image]

[Autolevel.zip](https://github.com/vlachoudis/bCNC/files/2973979/Autolevel.zip)

Harvie, 2019-03-16:
> I also noticed that the autolevel was reading the origin as about 0.013 mm higher than my probe for that loaction,

Due to deceleration, GRBL continues to move bit further after probe detected contact! This means CNC movement stops bit lower than probed point is. You can use "goto" button to go back to that point, or the checkbox to do that automaticaly:

[image]

Marcustuck, 2019-03-16:
Ok, that makes sence, thanks.

MARIOBASZ, 2019-03-16:
@Marcustuck, be careful with these errors if you work with the charged compensation matrix:
1- milling out of the scanned area
2- use G53 (this is awkward: instead of raising to Z safe, embed the tool in the PCB)

Marcustuck, 2019-03-17:
Thanks for your help, very happy with the results.
[image]

Need to work on my laser alignment...
[image]

Harvie, 2019-03-17:
Nice results... What EDA / CAM software do you use? What endmills?

Marcustuck, 2019-03-17:
I have been using Eagle but as I don't want to pay for another license (for the larger area on other projects) I have just started to try out KiCAD (its mush easier to adjust pad sizes etc, which is handy for this kind of prototyping).  I then use FlatCAM to produce the G-code.  I’ve only had the CNC for a short while, but I am impressed with it so far.  I am starting to learn what feed rates etc give the best results, hopefully I will receive some more cutting bits soon.  At present I only have a 0.2 mm 20 degree, but I have some 0.1 mm on order.  I am also hoping to get some double-sided copper boards too.

My first project was a limit switch interface for the Woodpecker board.  I made the interface using a quad opto-isolator and have the XYZ ad Probe go through it, so no noise issues.

I hope to be able to work up to bigger projects like my AFAM computer, I’ve had to send off to get this complexity circuit made in the past.

I’m actually driving around the world in my truck, so this is just a fun hobby, not what most travellers pack in their trucks 😊

Marcustuck, 2019-03-17:
I have been using Eagle but as I don't want to pay for another license (for the larger area on other projects) I have just started to try out KiCAD (its much easier to adjust pad sizes etc, which is handy for this kind of prototyping).  I then use FlatCAM to produce the G-code.  I’ve only had the CNC for a short while, but I am impressed with it so far.  I am starting to learn what feed rates etc give the best results, hopefully I will receive some more cutting bits soon.  At present I only have a 0.2 mm 20 degree, but I have some 0.1 mm on order.  I am also hoping to get some double-sided copper boards too.
My first project was a limit switch interface for the Woodpecker board.  I made the interface using a quad opto-isolator and have the XYZ ad Probe go through it, so no noise issues.

[image]
[image]
[image]
I hope to be able to work up to bigger projects like my AFAM computer, I’ve had to send off to get this complexity circuit made in the past. (full story http://www.tuckstruck.net/truck-and-kit/modifications-and-repairs/afam-computer-replacement/)
[image]
I’m actually driving around the world in my truck, so this is just a fun hobby, not what most travellers pack in their trucks 😊

closed 2019-03-17 by Marcustuck

MARIOBASZ, 2019-03-17:
I'm interested in isolating noise from career limits.
In eagle you can integrate pcbgcode to give you the gcode

Marcustuck, 2019-03-17:
I have attached my 'limit switch' design.  You will see there is a link wire between LSP1 and LSP2 if you want to use the Woodpeckers 5V supply.  If you want to use an external supply for the LEDs and opto-sisolators you can attach it to LSP1.  I've not seen any noise issues using this circuit, its been very reliable.  Its also nice having the LEDs to show when a limit switch is activated or the probe circuit made (I normally put something metal between the cutting bit and PCB before I start a probe to make sure the probe LED comes on to prove the connection is good).

I will look into "in eagle you can integrate pcbgcode to give you the gcode", thanks.

[Limit Switches.zip](https://github.com/vlachoudis/bCNC/files/2975454/Limit.Switches.zip)

Marcustuck, 2019-03-23:
Mariobasz,

Thanks for the tip about PCB G-code in Eagle.  I've loaded the ULP and I like the way it does the isolation routing rather than having to do a copper removal.

Thanks,

Marcus

massiveradiation, 2020-09-11:
Folks, what makes this thread very confusing is that the nomenclature is not consistent or clear. An "autolevel button" is referred to many times, but there is no "autolevel button." Are you talking about the button that has a target symbol on it, that is labeled "zero," which is in the Probe>autolevel area of the software? I honestly have no idea.
Please, please clarify.
Thank you,
Tyler