vlachoudis/bCNC #1194: Expand probe possibilities: Center

MARIOBASZ, 2019-03-15:
In addition to finding the center of a figure from the inside, it would be good:
  Interior or Exterior (Go in the direction of the center, when you touch it, go back, go up, go over the measurement, go down, look for the center until you feel it):
Vertical
Horizontal
Circle (current)

pomeroyb, 2019-03-25:
I just wanted to let people know that I am currently working on adding some of these routines, and have been drawing up icons to go with them.
[image]

I've already got the framework for simple edge probes, and am currently working out the best way to do more complex probes (like center of a boss). The simple way (mentioned by Mariobasz) is going up and over the part, but that requires the user to input the approximate size of the boss and the z safe height. There's a good chance for a crash if the user makes an error in these inputs

A safer, more automated (but probably slower) probing routine is what we see at the end of [this video](https://youtu.be/O9DjQqlitX0?t=468). The software incrementally creeps around the boss' edges to eventually find the center of the box. I imagine we could use the ALARM:5 flag from G38.2 and G38.4 for this, as long as bCNC has a mechanism for detecting and automatically bypassing this error. I haven't looked too hard at this yet, so I'm not sure if it's possible.

MARIOBASZ, 2019-03-25:
@pomeroyb, excellent! I am happy to always read those who get involved, think, propose, contribute! that helps the rest. And it generates a particular satisfaction in oneself to give some contribution. A hug.

MARIOBASZ, 2019-03-25:
following your idea, it occurs to me, there could be a routine that is anti-collision (maybe idea to discard, or validates for flat pieces)
1- Stand on the piece
2- Find z of the piece
3-Go up to safe parameter z
4- start probe routine

MARIOBASZ, 2019-03-25:
very ambitious idea?
if we have a camera, which can find edges, (or use a laser pointer), with artificial vision, could it be found, although it is approximately, the part's limits, and then a better-quality probe?

pomeroyb, 2019-03-25:
Computer vision and cameras are far outside the scope of what I will be adding -- I am only looking at probing routines for touch probes.

It looks like I should be able to automatically detect alarms and clear them using
`s.event_generate("<<AlarmClear>>")`

So unless anyone suggests otherwise, I will probably use that to implement the more advanced probing routines.

Harvie, 2019-04-02:
> I just wanted to let people know that I am currently working on adding some of these routines,

I would really like to convert probing framework to work in synchronous manner, so we can do more advanced probing. Currently the autolevel g-code is pre-generated and then just run as a whole. I would like to have possibility to modify the probing toolpath as it gets probed, so we can optimize on the fly. Also i would like to implement some kind of "probing plugins" on top of that synchronous probing API.

> and have been drawing up icons to go with them.

There is preferred style for the new icons: https://github.com/vlachoudis/bCNC/pull/1079#issuecomment-444853539

pomeroyb, 2019-04-05:
@Harvie I had not seen that style guide for the icons, but I think my icons are within spec. Here's a preview of what they currently look like: 

[image]

I did not make 16x16 icons, but opted for 32x32 (To mimic the "target32.gif" icon that is currently used for probing). Let me know if I need to change that.

Harvie, 2019-04-05:
I thought that you want to use this in the ribbon / tool selection, where we have 16px icons and this would break the layout. Using it on buttons like this is probably OK.

pomeroyb, 2019-04-10:
I'm almost done implementing the edge finding routines. These routines automatically adjust the current work position to zero on the center of the probe diameter.

I'm currently running into two issues:

1. You cannot have a "probe" command (G38.X) or a "set work position" command (G10 L20) as the last command in the sequence. I use a "move in machine coordinates" command (G53) to move the machine slightly. Grbl will throw an alarm if this is not used. That quirk took a while to track down! It's not a showstopper -- I just wanted to have that documented somewhere for future reference.

2. bCNC sends a reset to Grbl on job complete, which resets all modal groups, including the WCS. This means if you are probing in the G56 WCS, the probe will complete successfully, but you will be returned to the G54 WCS. Not a deal breaker, but a little annoying and could result in a crash if someone wasn't paying attention and moved back to their original WCS.

Neither of those issues block the functionality of the setup, however, and I'm hoping to have a pull request made on this relatively soon.

Harvie, 2019-04-14:
> This means if you are probing in the G56 WCS, the probe will complete successfully, but you will be returned to the G54 WCS.

Not good, we should fix this.

MARIOBASZ, 2019-04-15:
I agree, it is a serious mistake: this problem mentions it as an isue: if I work in g56 and squeeze stop, it changes to g54, great risk of crash! I do not remember, but I think the same thing happens at the end of the milling. It is a very dangerous behavior

klint-k, 2020-01-10:
I was watching this video https://youtu.be/bhZdbgM6S70. Would the routine allow for a plate like in the video?

MARIOBASZ, 2020-01-10:
I like that: load the parameters of the edge finder and a button to start the movement routine