vlachoudis/bCNC #1195: Tabs Issue

wheemzy1, 2019-03-20:
I'm new. I tried to import an .stl file but it says I need to go to Tools ----> Plugin

Problem is: I don't have a tools tab. 

The only tabs I can view are: File, Control, Probe, CAM, Editor

I have version 9.14-dev

I tried expanding the screen size, I have a 32" monitor so this should not be the issue. 

Please help.

Harvie, 2019-03-20:
"Tools" tab was recently renamed to "CAM".
That warning popup haven't reflected that yet...

closed 2019-03-20 by Harvie

MARIOBASZ, 2019-03-20:
@Wheemzy1, @Harvie, I never used slice mesh, I thought it was for 3D milling.
What does it do to import the contours in 3d? Then I guess you can mill (only with the option on) using trochoid 3d, I can add option "normal cut without trochoids" (trocoid diameter = tool diameter). I do not know how to make compensation for 3d tools and I do not have time to investigate.

MARIOBASZ, 2019-05-06:
@Harvie, I could not import .stl files, my error, from the file? can you send me some file that works to try it?
Another issue: Trochoid 3D can follow a 3d route, I would like to see if it can be applied.

Harvie, 2019-05-06:
@MARIOBASZ see #1223

Harvie, 2019-05-06:
These should work:

[sphere.stl.zip](https://github.com/vlachoudis/bCNC/files/3148789/sphere.stl.zip)
[chaos-cam.ply.zip](https://github.com/vlachoudis/bCNC/files/3148790/chaos-cam.ply.zip)
[chaos-cam.stl.zip](https://github.com/vlachoudis/bCNC/files/3148791/chaos-cam.stl.zip)

commit 22e67c18 (2019-03-20): Fixed #1195