vlachoudis/bCNC #1196: Huanyang VFD

ThomTrab, 2019-03-20:
Hello,

I would like to control a Huanyang VFD by RS485 through bCNC, but I don't know where to start.

I'm ok writting a phyton module for it but I have no clue how to interface it with the software.

Thank you for your help

LYHTSPD, 2019-03-20:
Is this what you are looking for?
http://linuxcnc.org/docs/html/man/man1/hy_vfd.1.html

Harvie, 2019-03-20:
If you will make python class to communicate with RS485 for both control and status report (rpm, voltage, current,...) i will help you to integrate it to bCNC...

ThomTrab, 2019-03-21:
@LYHTSPD, yes that what I'm looking but for bCNC

@Harvie, I have already found this which should do the heavy lifting to communicate with the VFD through RS485
https://github.com/drogenlied/pyHYControl

I think it only need some sort of wrapper around it to have it integrated in bCNC and that where I need help. Where this should be plugged in bCNC ?

Harvie, 2019-03-22:
> Where this should be plugged in bCNC ?

I will have to take a look. I need to figure out how to make it unobtrusive for people who don't use it.

lolsifer, 2019-03-22:
@ThomTrab, I have recently set up a router that works like this...

Not sure if the best option is integrating into bCNC, I did it by linking a second Arduino board to the GRBL board via SPI... then I use a MAX485 Arduino module from this second board to talk to the VFD... I've had noise issues which I've added messy workarounds for (not really going to commit it to the community in it's current state)

I'm probably going to modify it into my GRBL board and put the MAX485 on the GRBL board, haven't had the time to learn my way around GRBL yet, don't really want to add interrupts which could cause steppers to lose steps due to communication with the VFD.

Anyway; following.

Harvie, 2019-03-22:
I've been considering possibility to use unmodified GRBL with second arduino board reading it's PWM output and converting it to RS485... Arduino has library for decoding PWM back to numeric value. Biggest downside of this solution is that it will not allow you to read spindle state to computer.

Also it might be easier to wire PWM directly to VFD...

ThomTrab, 2019-03-22:
I'm sorry, I think haven't clearly explain what I was looking to do.

I have a USB/RS485 converter which will do the connection between the PC and the VFD. With the Python lib I already shared, I have a way to communicate to the VFD from the PC. What I was looking for is a way to get motor related action from bCNC and reroute them to the VFD through the USB/RS484. By using a dedicated converter, I will also be able to get status from the VFD. 

That's the setup required for LinuxCNC (I'm moving from LinuxCNC to bCNC).

Hope I have better explained my thought :wink:

Harvie, 2019-03-22:
> I'm sorry, I think haven't clearly explain what I was looking to do.

You did, but we suggest there are also cheap hardware solutions to your problem. They are interesting, because they will not be dependent on used software. In GRBL world the arduino is supposed to control the spindle rather than computer. Especialy because controling spindle from bCNC will not be able to sync spindle with g-code, because GRBL caches ~10 g-code lines, so bCNC is 10 lines ahead. This is huge problem.

paulvdhoeven, 2019-04-23:
I also have a OneHungLow Inverter and while RS485 may have some advantages, I did not find RPM accuracy too important. Note that the RPM Will never be acurate anyway because the inverter controls and induction motor, and these always have slip.

I had a look at some example circuits, and they did not look very promising, so I designed my own Isolated PWM -> DC circuit.
[image]

It works as follows:
The "primary side" has 4 LED's in series, and this means that there is no current through them if "Spindle_PWM" is not connected. If Spindle_PWM has a PWM signal, ten alternately the top and bottom optocoupler is turned on. This makes the circuit pretty lineair.

On the secondary side. One of the opto couplers acitively puls the RC circuit high or low, again for linearity. There are 2 resistors in series (R504 and R507)  These are to protect the optocouplers in case they are both turned on inadvertently.

The optocouplers are ot strictly necassary because the 10V interface of the Frequency inverter is already isolated, but I needed to amplify the PWM from 5V to 10V anyway, and optocouplers are just as cheap as transistors. I built this circuit (together with direcion swithing) on a perf board and put it in the frequency inverter itself to keep the analog signal paths short.

Harvie, 2019-04-24:
@paulvdhoeven i was able to get huangyang VFD working with even simpler circuit... I just used single optocoupler between arduino and VFD. No additional components. VFD has configurable filter (PD071), so you don't need any capacitors to smooth the PWM, just enable that in menu. Also there was no need to convert PWM to voltage, since you can setup VFD to be driven by current rather than voltage (see PD070 in huangyang manual). Last but not least i don't usualy use resistors when driving LEDs by Arduino. I know it's not reccomended, but over the years i didn't have single arduino or led failed on me because of this. I just connect LED directly between arduino input and GND. Never had any issues.

Only problem is, that you need two optocouplers. One for setting frequency and one for starting the spindle. VFD can't be configured in a way that PWM signal alone will cause it to run. Probably for safety reasons. You don't want some random peak of noise on the line enabling your spindle.

Harvie, 2020-11-23:
Now there seems to be quite elegant solution integrating Huanyang RS485 communication directly to GRBL, i would be happy if you can help with testing:
https://github.com/terjeio/grblHAL/issues/68

ThomTrab, 2020-11-23:
I would be glad to it to a test but since then we moved away from GRBL for our machine due to unsolved electromagnetic issue in the wiring. We took advantage of software filtering capability of LinuxCNC to solve our issue.