vlachoudis/bCNC #1197: Auto level problem

atlask1, 2019-03-22:
Hi, this is not a really issue, is an help request to understand the correct way for milling right after auto level.
I'm starting with a cnc with atmega board . 
The file is generated from eagle and gcode ulp. 
I try to follow the tutorial and the wiki, but I do not understand where I'm wrong.
My question is:
after a correct probe sequence, wich is the value I see on canvass that I have to consider for 'Zero? Higher? Lowest? Middle?
I notice that there is a lot of difference between some value, and I know it's a copper surface problem, but maybe the calibration sequence that I follow is incorrect.
Maybe there's some configuration in gcode that I have to change, but I'm new on this and I need some help. 
Have someone a link to some simple tutorial to follow? Or some indication more specific for the autolevel? 
thanks in advance

Harvie, 2019-03-22:
> after a correct probe sequence, wich is the value I see on canvass that I have to consider for 'Zero? Higher? Lowest? Middle?

There is button to set which one is the zero. Any time you can change it as needed.

atlask1, 2019-03-22:
Thanks for your answer, I know the Zero button, you intend I use it any time during milling? I believe that I have to choose a value that I consider zero and autolevel modify other z value with the difference between the zero and the other value, so I must choose for example the lowest, and the program do a calibration of other points. 
Could you be more specific in a correct sequence? Thanks a lot, it’s my first works  with , try to understand me!!

Harvie, 2019-03-22:
After autoloveling you jog to your desired XY zero. Perform Z touchoff to top of your stock surface. Then you zero XYZ coordinates. Then you zero your autolevel data. This will modify autolevel data to be zero at current XY location:

[image]

closed 2019-03-22 by Harvie

atlask1, 2019-03-22:
Yes , the ‘Zero ‘ button is the same.
My question is: which is the best XY zero coordinate? You say ‘your desired XY zero’ , but I don’t think is the same, I believe I have to choose one particular point or is not important? 
After select a Zero xy I performa a Z touchoff, but after this , if I zero the xyz coordinate, when I start milling, the working area is wrong, because the first zeroing point is different, if I set zero only Z, also after Zero button , I see the Z axis move in respect of autolevel, but milling is not correct , more or less deep.
Sorry, don’t consider me ‘stupid’ 😊, but is not clear step by step auto level process, switching from control window to probe window , set initial zero calibration, set autolevel and set zero for autolevel. 
I'm getting used to cnc and bCNC sw, I consider this part very important and I something is not clear to me, I must ask. 
If you prefer link me to other documentation or tutorial, feel free, but I think I’m turning around my stupid not understanding.
Thanks

Harvie, 2019-03-22:
> which is the best XY zero coordinate? 

your g-code has origin (zero). you have to set your zero on your stock in the place where you want your milling to be done... your autolevel XY zero should be set in place, where you've done the Z touchoff obviously, otherwise it does not make sense... In many cases this can be the same point as your coordinate zero...