vlachoudis/bCNC #1198: DXF Inside profile appears on outside, pocket crashes.

DaveDavenport, 2019-03-23:
Version: 2a6e00d6783e6277e33b43274758402986f5d014

The two path if you try to do a profile on the inside, it appears on the outside.
[image]

Then inside:
[image]

So tools like pocket, go completely bonkers (crash or this result).

[image]

I am not sure what is wrong (dxf) but in anyway the tool should not crash if something is wrong.

[vb-bugreport.zip](https://github.com/vlachoudis/bCNC/files/2999290/vb-bugreport.zip)

Please let me know if you need more info.

Mill bit diameter is 1.5mm.

Harvie, 2019-03-23:
From what i remember this happens when endmill diameter is very close to pocket width. That might be your case, but i haven'ŧ checked yet...

DaveDavenport, 2019-03-24:
It is, but I also tried this with 1mm end mill setup and even 0.1mm endmill. still happened.

In the end I managed to mill these two by selecting an outside profile and milling that.

MARIOBASZ, 2019-03-24:
if the internal fressdo does it on the outside, and the external milling on the inside, it is that the route is not closed: look what happens if you make a straight line or an "L"

DaveDavenport, 2019-03-24:
I tried it with/without the 'close path' option in bcnc.
Initially the path was indeed not closed, that caused other crashes (regrettable).

MARIOBASZ, 2019-03-24:
there is a serious error for pocket if there is no space for the tool's diameter: unfortunately hsy qye restart .....

DaveDavenport, 2019-03-24:
I must say I *love* bcnc and use it a lot (parts milled using bCNC have gone up to (almost) space). 

I do think it is a big problem that if something goes wrong, its often very unclear of what is going wrong. 
(I mostly get a backtrace about non-itterable object or so). 

Then I have to restart, set everything up again (As settings are lost) and then slowly try to pin-point what is going wrong. (already found that making my tool a not nicely round number like  1.499 mm instead of 1.5 helps often helps (if I don't mind the diff), but still)

This process often causes me 1-2 hour delay on starting the job. Something that could validate gcode /dxf for possible issues would be great. Its hard to find issues in provided dxf files you have not found yourself. Maybe validation tools exist already?

MARIOBASZ, 2019-03-24:
I agree, it is frustrating when you must restart for any inconvenience (pocket is one for bcnc, alarm for exceeding soft limits or autolevel I think it is inconvenient for grbl), really a serious and antipatico inconvenience. It would be optimal to solve this problem. Practically all the objects I'm doing are designed and modified with bcnc.
Curiosity, can you show some of the things that fly almost to space?

Harvie, 2019-03-24:
> parts milled using bCNC have gone up to (almost) space

What kind of parts are you making? Stratospheric baloons? :)

> I do think it is a big problem that if something goes wrong, its often very unclear of what is going wrong.

Well obviously it's a bug, not intended feature.

> (already found that making my tool a not nicely round number like 1.499 mm instead of 1.5 helps often (if I don't mind the diff), but still)

I have some plans how to fix profile/pocket problems. But time is the limiting factor. Since everyone including me is doing it for free.

Anyway. I've never run into this problem in my project. I usualy choose smaller endmill for pocketing. Also when i am going to drill or groove feature that is exact size of the milling bit, i usualy just draw center point or centerline in the CAD and use cut/drill operation directly instead of making profile.

> Then I have to restart, set everything up again (As settings are lost)

Actualy i was thinking that it might be possible to make bCNC parametric like freecad. So bCNC will save your "intention to do profile" of shape. rather than profile g-code itself. This has pros and cons. And might take a lot of time to implement in useful manner.

> Something that could validate gcode /dxf for possible issues would be great. Its hard to find issues in provided dxf files you have not created yourself. Maybe validation tools exist already?

What kind of issues are you talking about? bCNC has tools to round the g-code precision and to close open paths (as long as the path is open only in single spot). Also we have arcfit plugin to simplify subdivided g-code. But it's kinda hard to automaticaly figure out something useful from broken file...

DaveDavenport, 2019-03-24:
> I have some plans how to fix profile/pocket problems. But time is the limiting factor. Since everyone including me is doing it for free.
> 

Know the feeling, writing (and written) lot of opensource software.
Not my intention to pressure anybody. As I said, I love bCNC and am very grateful for it and everybody who worked on it.

DaveDavenport, 2019-03-25:
@Harvie, @MARIOBASZ  I helped making some frames for use in one of the student projects on the [rexus](http://rexusbexus.net/rexus/) rocket (https://www.sscspace.com/rexus-25-26/).

Harvie, 2021-02-23:
@DaveDavenport Can you please try to process your shape with new tool called "offset" which is now available in latest pip release of bCNC?
Old "pocket" and "profile" tools are now obsolete and will be removed in future releases in favor of the "offset" tool.

DaveDavenport, 2021-02-23:
I will put it on my todo list! will take a bit of time for me getting around to it.

Harvie, 2021-02-25:
> I will put it on my todo list! will take a bit of time for me getting around to it.

i am going to release 0.9.15 soon, so if you manage to test it on time, you can be sure everything works as expected in new release.