vlachoudis/bCNC #1199: Drag Knife post processor not working with some toolpaths

Rhysun, 2019-03-24:
The drag Knife post processor throws a math domain error with one of my gCode files. But if I enclose...

`angle = degrees(acos(TA.dot(TB)))`

in a `try: except` block, thus...

```
try:
    angle = degrees(acos(TA.dot(TB)))
except:
    continue
```

... the problem goes away. I have absolutely no idea what is going on with this toolpath so the above solution is probably a bit 'hacky'. Anyway, here is the problematic file:

[yrOrsaf3.ngc.zip](https://github.com/vlachoudis/bCNC/files/2999939/yrOrsaf3.ngc.zip)

Harvie, 2019-03-24:
There was error in math library, fixed that. Thanks for reporting.

closed 2019-03-24 by Harvie

Harvie, 2019-03-24:
BTW would you mind doing some pictures or youtube video about your bCNC dragknife use? I'd love to see how this is usefull to other people and maybe promote the feature a bit :-)

MARIOBASZ, 2019-03-24:
could you add a site on github or on wiki or slgo, to upload examples, tips, trivia, phytos, videos, experiences, curiosities of bcnc / plugins?

Harvie, 2019-03-24:
@MARIOBASZ everybody can edit bCNC wiki. Go ahead if you wish. I am going to make some tutorial videos after 0.9.15 release.

commit 645d10f8 (2019-03-24): Float precision error in dot product method was fixed. Fixes #1199