vlachoudis/bCNC #1200: CAM/Scaling replaces M commands

smooker, 2019-03-24:
Replaces M04 GRBL laser command with M03 during scale.

Harvie, 2019-03-24:
@MARIOBASZ any ideas?

smooker, 2019-03-24:
Additional info:
vim: build/lib/bCNC/plugins/scaling.py 209,24-59

```
                                        bid_block.append("(---- Scale (x "+str(xscale)+" : 1.0),(y "+str(yscale)+" : 1.0),(z "+str(zscale)+" : 1.0) ---- )")
                                        bid_block.append("(center "+str(center[0])+" ,"+str(center[1])+" )")
                                        bid_block.append("M03")
                                        bid_block.append("S "+str(rpm))
                                        bid_block.append(CNC.zsafe())
                                        bid_block.append("F "+str(zfeed))
                                        bid_block.append("g0 x "+str(info[0])+" y "+str(info[1]))
```

CNC.zsafe() is not needed here for laser engraving. (milling of scaled model too :) slows down the process or breaks the instruments with coefficients bigger than 1.000)
A lot of lower cases and spaces.

Regards,
smooker.

MARIOBASZ, 2019-03-24:
Never use laser.
I will add Laser option.
What should be on the Gcode replacing Zsafe?
Must say M03 activate, M05 deactivate?
It is easy to modify in the plugin.

MARIOBASZ, 2019-03-24:
& Harvie, Laser option is registered as variable in bcnc? If it is correct, you tell me how to summon it thus saving time, and it is not necessary to place the option in the plugin?

smooker, 2019-03-24:
M03 or M04 in front of every path (block)
M05 to turn off the laser in the end of the path (to not make scratches between bridges :) )

MARIOBASZ, 2019-03-24:
M03 or M04 in front of every path (block)
M05 to turn off the laser in the end of the path (to not make scratches between bridges :) )
Ok

MARIOBASZ, 2019-03-24:
@smooker, I did not understand what are you saying:

I think that Zsafe should be scaled too (in milling mode) and ingnored in laser.
Opinions ?

MARIOBASZ, 2019-03-24:
place the laser / milling option
if there is a "laser variable" stored in bcnc, I will try to call it (@Harvie, delete an earlier comment by mistake, can you tell me if it is stored in bcnc as a variable, and how do I call a laser variable to save time?)
If it does not exist, in the plugin I will place the Laser / engraving option

Harvie, 2019-03-24:
Scaling plugin should only change the scale of g-code coordinates. I don't understand why it should change how motors/lasers are engaged...

MARIOBASZ, 2019-03-24:
I'm not in my machine.
but at the end of the block must go to zsafe for safety in milling, and for laser they say "change between m03 / m05)

MARIOBASZ, 2019-03-24:
@smooker, if you send me a simple code to burn a square with laser, or two squares, one in each block, I will understand faster

smooker, 2019-03-24:
just a minute

MARIOBASZ, 2019-03-24:
@Harvie, when you do laser pocket operation, in more than one figure, write m03m05 instead of zsafe ztarget?

MARIOBASZ, 2019-03-24:
@smooker, I'm glad you can use scaling.
here a use to scaling and modified (even in test) for 3d, oval from curculus.
Disadvantage: My wife told me "Mario, go and take a shower"
[image]

smooker, 2019-03-24:
smooker@sh21 /mnt/vg3.2/home/smooker/src/bCNC/examples $ cat MARIOBASZ.ngc 
```
G21 (All units in mm)

S1000 (speed of the motor is pwm to laser)

G00 X0.0 Y0.0 Z0.0 (laser stays on Z=0 always)
M04 (turn on grbl last in "auto" mode)
G01 X0.0 Y10.0 Z0.000000 F400.000000
G01 X10.0 Y10.0 Z0.000000
G01 X10.0 Y0.0 Z0.000000
G01 X0.0 Y0.0 Z0.000000
M05

G00 X20.0 Y0.0
G91
M04 (turn on grbl last in "auto" mode)
G01 X0.0 Y10.0 Z0.000000 F400.000000
G01 X10.0 Y0.0 Z0.000000
G01 X0.0 Y-10.0 Z0.000000
G01 X-10.0 Y0.0 Z0.000000
M05
```

MARIOBASZ, 2019-03-24:
Thank you!
Have you corrected the plugin? Do you want to make your modification? Do you want me to do it?

smooker, 2019-03-24:
I've tried, but w/o success. Have no idea of python :)
It's enough just to bypass the M commands as they are and as Harvie said :)

smooker, 2019-03-24:
[image]
[image]
[image]

MARIOBASZ, 2019-03-24:
it is that it selects the route, and on it realizes scale, I think that it ignores the line that says m04 m05, it should fix me. At least I will put "select for laser" if it is simpler, until final solution.
One question, for when installing a laser: ignore height z, or if you write z-1 low z?

smooker, 2019-03-24:
Do not ignore Z, because some people may use it for focus/defocus (increase/decrease the area of the "tool").
Just scale Z too. It'll be up to the user to put the correct scale factor.

MARIOBASZ, 2019-03-24:
with laser you use pocket operation? automatically type m04 m05?

smooker, 2019-03-24:
yes. it's not needed to turn off the laser with M05 for deeper engraving (pocket), but it's ok.

MARIOBASZ, 2019-03-24:
I see that you are trying to modify the plugin, Animo! Believe me I barely got some python!
experience modifying what I wrote, surely you will be able to do it, but between today and tomorrow I modify it so that it serves you and you do not have to correct the gcode by hand

MARIOBASZ, 2019-03-24:
Place boolean laser field (look at trochoid 3d plugin or helical, I think there is a Boolean field)
At the end you must say Zsafe, for no laser and m05 for laser

MARIOBASZ, 2019-03-25:
How BCNC behaves profile, pocket or cut, to know if it is with laser or with milling? (write M03 M05 or write Z safe or Z objective?)
Is there any laser / cutter option?

MARIOBASZ, 2019-03-25:
I have added an option for Laser scaling (as long as you do not know how bcnc knows that you should write for laser, it is a solution)
@smooker, can you check that everything works correctly?

smooker, 2019-03-29:
sorry for the delay. will check later and post the result

smooker, 2019-03-30:
I can not see any difference :)
Which one is the commit ?

MARIOBASZ, 2019-03-30:
[scaling.zip](https://github.com/vlachoudis/bCNC/files/3025746/scaling.zip)

smooker, 2019-03-30:
[image]
Just a suggestion...