vlachoudis/bCNC #1201: Binary_STL_Writer

smooker, 2019-03-25:
```
[smooker@smoolap ~/src/bCNC]$ ./bCNC.sh 
Launching bCNC from .
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/smooker/src/bCNC/bCNC/__main__.py", line 57, in <module>
    from CNC import WAIT, CNC, GCode
  File "/home/smooker/src/bCNC/CNC.py", line 16, in <module>
    import pickle
ImportError: cannot import name Binary_STL_Writer

```

smooker, 2019-03-25:
update-alternatives to python2 and repull fixed it.

closed 2019-03-25 by smooker

Harvie, 2019-03-26:
See wiki on how to properly install and run bCNC. bCNC.sh is legacy compatibility script.

nvdl, 2019-08-03:
This error occurred on my side due to stale ".pyc" files after doing "git pull" after almost a year.
Since some ".py" files have moved, clean the stale ".pyc" files as below.

find -type f -iname *.pyc | xargs rm