vlachoudis/bCNC #1203: Problem to see command line

MARIOBASZ, 2019-03-26:
Download the latest version, the panel looks just bigger in verical, and to write on the command line (you can see less than half its height), I must hide the panel of control keys.

Harvie, 2019-03-27:
What do you mean? Send pictures.

MARIOBASZ, 2019-03-28:
In the rest of the tabs, the command line menu box is displayed correctly. The inconvenue is in the Control tab: you can not read what is written (1/2 of the text box)
[image]
[image]