vlachoudis/bCNC #1204: Doing "cut" on simple non-closed path will close the path on the bottoming pass

MrSurly, 2019-03-27:
Doing a simple cut of a non-closed path from an SVG:

[image]

Gives unexpected results in Camotics -- essentially the path is inadvertently closed just before the final bottoming pass:

[image]

Looking at [CNC.py:3532](https://github.com/vlachoudis/bCNC/blob/master/bCNC/CNC.py#L3532): `if splitpass: path.invert() #Fixes split ramp case`, commenting out this line fixes the issue.  

I _think_ that since `path.invert()` is called at [line 3515](https://github.com/vlachoudis/bCNC/blob/master/bCNC/CNC.py#L3515) _and_ [line 3518](https://github.com/vlachoudis/bCNC/blob/master/bCNC/CNC.py#L3518) before entering the bottoming block, then this final `path.invert()` shouldn't be necessary, but I'm not entirely sure.

Harvie, 2019-03-27:
Unfortunately this happend to me out few days ago while doing actual cut. But wasn't sure what happend. Thanks for catching this.

MrSurly, 2019-03-27:
No problem, would have submitted a PR, but I'm not familiar enough with the code to know for sure if removing line 3532 would cause other problems.

Harvie, 2019-04-02:
Don't worry, i'll soon take a look myself...

Harvie, 2019-05-02:
> I'm not familiar enough with the code to know for sure if removing line 3532 would cause other problems.

This was actually the case, but i've managed to untangle the problem and it should be OK in the latest git version...

commit a6e3538f (2019-05-02): Improved cutting logic. Fixed #1204
closed 2019-05-02 by Harvie