vlachoudis/bCNC #1205: Scrolling IDLE Message

barnold96, 2019-03-27:
I've read other posts explaining the issue I have, but none provide a solution.  My RPi3 is attached to an Arduino UNO via a USB port.  All functions seem to work during testing. I can jog all three axes OK.  I cannot see the parameters I need to adjust due to the Terminal constantly scrolling an IDLE message.  When I send $$ to view settings, the settings pop on screen but are immediately scrolled up and off screen.  The same happens with any other grbl command I enter.  How do I stop this?

Harvie, 2019-03-27:
Can you please attach some screenshots of the problem?

barnold96, 2019-03-27:
For some reason, I can't get a screen print.  I don't have any problem hitting 'Print Screen' and recording other windows, but when I try the same on the VNC window, nothing happens.

I'm using bCNC v0.9.14-dev with grbl 1.1f on the Arduino.

This is what is scrolling in the Terminal window.
<Idle,MPos:0.0000,0.0000,0.0000,WPos:-4.0000,1.0000,3.0000>
<Idle,MPos:0.0000,0.0000,0.0000,WPos:-4.0000,1.0000,3.0000>
<Idle,MPos:0.0000,0.0000,0.0000,WPos:-4.0000,1.0000,3.0000>
<Idle,MPos:0.0000,0.0000,0.0000,WPos:-4.0000,1.0000,3.0000>
<Idle,MPos:0.0000,0.0000,0.0000,WPos:-4.0000,1.0000,3.0000>
<Idle,MPos:0.0000,0.0000,0.0000,WPos:-4.0000,1.0000,3.0000>
<Idle,MPos:0.0000,0.0000,0.0000,WPos:-4.0000,1.0000,3.0000>
<Idle,MPos:0.0000,0.0000,0.0000,WPos:-4.0000,1.0000,3.0000>
<Idle,MPos:0.0000,0.0000,0.0000,WPos:-4.0000,1.0000,3.0000>

Harvie, 2019-03-27:
Try experimenting with Status report, mask setting of GRBL ($10= )

barnold96, 2019-03-27:
That fixed it!  I entered $10=1 and the scrolling stopped.  I re-created the issue by entering $10=3.  The Arduino and attached Break-Out-Board were purchased from a merchant who must have set the $ parameters.

Thanks for your time!

Harvie, 2019-04-02:
This is very strange... `$10=3` on GRBL v1.1f (atmega328p) from github.com/gnea/grbl works perfectly for me with latest git master of bCNC...

Harvie, 2019-04-02:
On the other hand it does NOT work for me with `$10=2` and `$10=0`

Harvie, 2019-04-02:
I've asked GRBL developers about your problem... For me it looks like you have some unofficial or unstable development version. But let's see what they will tell us: https://github.com/gnea/grbl/issues/635

Harvie, 2019-04-02:
According to GRBL developers your device only claims to be grbl 1.1f, but it is not.

closed 2021-02-23 by Harvie