vlachoudis/bCNC #1206: Improve sampling in auto-level

jbvioix, 2019-03-29:
Wegstr has implemented auto-level to their machines. As you can see on the video (https://www.youtube.com/watch?v=-QqWOIUeAV4 near 3:00) auto level seems to use an irregular pattern not a regular grid. Maybe inflexion points on the tracks can be uses ?

MARIOBASZ, 2019-06-21:
@jbvioix , It seems that you are using a concept like the driller plugin to generate the autollevel points. I do not know if this results in more or less time of scanning the surface.

Harvie, 2019-06-29:
@jbvioix video link does not work...

carl1961, 2019-06-29:
Harvie, if the circle board has a circle edge.nc could or would autolevel use that as a reference?

MARIOBASZ, 2019-06-30:
Using a spiral path for test points can help?

Harvie, 2019-06-30:
@carl1961 yes. that is the plan, we just need to implement it :-)

jbvioix, 2019-07-04:
The video is https://www.youtube.com/watch?v=Mxa6xQFCyXA near 3:10 you can see the movement of the drill is not on a regular grid.

MARIOBASZ, 2019-07-04:
One strategy that I can think of to accelerate the sampling, is to discard the points that are far from the traces, if the time of intersection processing allows it

MARIOBASZ, 2019-07-04:
@jbvioix , By adapting driller, you can sample the routes at predetermined distances. It would be necessary to discard the points corresponding to nearby traces, it could be a good strategy

MARIOBASZ, 2019-07-04:
Maybe following this thread:
of something like pirograph, or a raster for laser:
From this:
[image]
to this
[image]

MARIOBASZ, 2019-07-11:
@jbvioix , I have seen it.
Do you think that strategy is an advantage?
[image]
try scan as driller plugin, combining anchor with intermediate points ?
(it is not a good representation)
[image]
[image]

MARIOBASZ, 2019-07-11:
I think a good strategy is the current grid, with the condition "only if there are traces at close range"

Donnyp751, 2019-09-06:
How WEGSTR auto levels is a very poor way of doing it. Not good for circuit boards. I have one of their machines. It literally just probes every N distance along the path of the gcode for a z height, then uses that as an offset. It changes the Z height when it reaches where it probed. It's not a very smooth transition. Causes sudden Z changes making it not isolate through some areas if the surface is at all curved.

Danivas13, 2020-12-15:
I think that the speed can be increase modify de Z safe. Generate a new z safe for probbing just a 0.3mm above the last probe point and then move to the next point.

Harvie, 2020-12-15:
> Generate a new z safe for probbing just a 0.3mm above the last probe point and then move to the next point.

We are not using probe just for flat surfaces autoleveling. We use it to scan 3D shapes as well. In such case the height can vary rapidly between the points.

Danivas13, 2020-12-15:
> > Generate a new z safe for probbing just a 0.3mm above the last probe point and then move to the next point.
> 
> We are not using probe just for flat surfaces autoleveling. We use it to scan 3D shapes as well. In such case the height can vary rapidly between the points.

Yes. but you can especify de max in the probe settings, the only thing that change es where start to count (origin Z-0 or the preview probe-Z.