vlachoudis/bCNC #1207: Z Axis work position coordinates randomly changing after job ends

EducatingSavvas, 2019-04-02:
I'm having an issue which I thought I'd raise here as I'm not sure what's causing it. At the end of a cutting job and after the tool returns to the origin position the z height work position coordinates randomly changes. The coordinate changes are not consistent and this effects performing repeat cut - say with a new piece held in a cradle. After this happens I have to move to the probe location and conduct a probe cycle and G10 command. 

I'm using bCNC with a self built CNC machine using the phoenix CNC controller. I'm using an old version of artCAM 2015 to create the g-code. I'm wondering if anyone has a similar issue and if this is a software / firmware or hardware issue?

Thanks in advance.

EducatingSavvas, 2019-04-03:
Just to add I've done some trouble shooting - disconnecting the VFD and other components but the behaviour still occurs. I also tired a different controller that I had started building and the behaviour did not occur which makes me think it's my controller. I will get in touch with it's maker.

EducatingSavvas, 2019-04-22:
I've figured out and thus answered my own question. G92 Position Register / Set Position commands will reset if a M30 command is executed at the end of a g-code job, or if the GRBL soft stop is pressed. This will affect re-commencing a jobs or starting a new one without re-setting the tool length. Instead of G92 it is recommended to use a G10 Set Tool Offset / Programmable Data Input command instead and that is the reason why. Nothing wrong with bCNC (or my CNC machine control box) - they works great.

Harvie, 2019-05-02:
Have you seen #1208 ? Maybe it's related...