vlachoudis/bCNC #1208: Danger: If work is interrupted, restart in G54

MARIOBASZ, 2019-04-03:
If I stop to interrupt the work, you start at G54 instead of maintaining the work site. While working on G53 and interrupting to correct height, I almost ruined the piece: Restart in coordinates G54

JipeB, 2024-05-04:
Hello,
I proposed a solution: [https://github.com/vlachoudis/bCNC/discussions/1838](url)

The workspace is detected when the program loads and validated automatically. This improvement of the code allows, among other things, to use the predetermined workspaces which have been stored in grbl.
Associated with a tool change by tool length offset, machining can be started directly without having to make the part origin.

MARIOBASZ, 2024-05-04:
Hi, how are you?Thank you. I don't know if the dro update has been implemented, but it is one of the points that are desirable to resolve.Hug.