vlachoudis/bCNC #1211: Installing version 0.9.15 - failing!

micksulley, 2019-04-09:
I am trying to install 0.9.15 on a Raspberry Pi, following the instructions on the wiki.  It first failed installing Pillow, but I fixed that by installing libjpeg-dev
The installation now completes without any errors, but fails when I run it.
Here is the install
`pi@pi-ox:~ $ pip install bCNC
Collecting bCNC
Collecting numpy>=1.12 (from bCNC)
Collecting pyserial<=3.0.1 (from bCNC)
Collecting Pillow>=4.0 (from bCNC)
Installing collected packages: numpy, pyserial, Pillow, bCNC
Successfully installed Pillow-6.0.0 bCNC-0.9.14.57 numpy-1.16.2 pyserial-3.0.1
`
Here is when I try to run it

`pi@pi-ox:~ $ python -m bCNC
new-config Utils <ConfigParser.ConfigParser instance at 0x75f497d8>
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/pi/.local/lib/python2.7/site-packages/bCNC/__main__.py", line 2675, in <module>
    main()
  File "/home/pi/.local/lib/python2.7/site-packages/bCNC/__main__.py", line 2542, in main
    tk = Tk()
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1823, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: no display name and no $DISPLAY environment variable
pi@pi-ox:~ $ 
`
Any ideas how to fix it?  I was running a previous version successfully.

Thanks
Mick

Harvie, 2019-04-09:
What version of python do you have?
Have you tried `pip2 install bCNC` and `python2 -m bCNC` ?

Harvie, 2019-04-09:
`no display name and no $DISPLAY environment variable `

This seems to me that you are not running it under X server. Are you trying to run it over ssh? It's GUI program which needs X server.

closed 2019-04-09 by Harvie