vlachoudis/bCNC #1213: Homing Cycle

arnold-b, 2019-04-09:
Hello, 
the homing cycle on my machine is x,y,z but i want z first. with the current cycle the tools can easily crash.  how can i change this. thanks

pomeroyb, 2019-04-10:
Are you using Grbl? The default homing cycle is defined [here](https://github.com/gnea/grbl/wiki/Set-up-the-Homing-Cycle#homing-cycle-steps).

If you are using grbl and Z is not homing first, you either have a custom firmware installed, or your stepper motor wiring is switched around (You have the X motor plugged into Z, for example).

arnold-b, 2019-04-10:
Thanks. I checked grbl. Z Homing was on cycle 3

closed 2019-04-10 by arnold-b

Harvie, 2019-04-14:
Homing cycle is handled by GRBL, not by bCNC
https://github.com/gnea/grbl/wiki/Set-up-the-Homing-Cycle#homing-cycle-steps