vlachoudis/bCNC #1214: Displayed tool movement area does not match work area (Autolevel Issue)

steppel, 2019-04-10:
Hi, 

I'm super new and wenn through everything I could find here and google to address the issue. I'm out of ideas and tutorials now. The tool head is displayed and is moving only in a fraction of the work area. So autolevel writes all the values almost on one point. I created the code files with flatcam and without autolevel all dimension match. I can set the dimension in the Probe section and I run everything. The only problem as mentioned is the tool does not move over the whole object, it stays pretty close to xy in a tiny square. I tried to find setting where I could possibly set that but no luck. Please have a look at the screenshots. My concern is that the autolevel data is incorrect since it looks like it's not the same dimension as the real object.

[image]
[image]

steppel, 2019-04-13:
Can someone please give me a clue? I tried the same now on a windows system. Obviously I seem to make a mistake here and haven’t the correct settings set up. The area where the tool head is displayed is a fraction of the object to scan. The head on the CNC runs correctly it just doesn’t display correctly in bCNC and hence writes all the auto level values on a tiny spot.

steppel, 2019-04-15:
Ok after hours and hours I figured that $13=1 was set. I set everything at the beginning to millimeter as well as $13 .. but looks like that was never written. setting 13 to 0 (so report in mm) fixed it. Hope that helps someone with the same issue

closed 2019-04-15 by steppel

Harvie, 2019-04-15:
Mentioned this in README