vlachoudis/bCNC #1215: Autoleveling question

micksulley, 2019-04-10:
I am trying to cut a PCB and I'm not sure that I am following the correct sequence for auto levelling
I have several operation -
- Drill holes for pins
- Cut tracks.
- Drill mount holes
- Cut outline

Autoleveling is only an issue for cutting tracks, so what I am doing is -

- Load cut tracks code
- Set margins, perform autolevel and zero
- Load pin holes code, it prompts me to save autolevel data and I say yes,  then run pinhole code
- Load cut tracks code, asks if I want to use autolevel date, say yes and run
- Load mount holes code and run
- Load outline code and run.

There is mention in the wiki of a .probe file, after I have performed the autolevel and saved it I see that tracks.gcode has been updated, but there is no .probe file, is that correct?
Also I am unsure about the warning in the wiki that it may be applied twice.  What is written is fine if I used autolevel with the cut track code and then executed it, but I have to load the pin holes code and run that first.
Any clarification would be most welcome

Mick

Harvie, 2019-04-14:
Do not autolevel the gcode file. It will get autoleveled automaticaly when sending to grbl. If you manualy apply the autolevel data to g-code it will get applied twice.

micksulley, 2019-04-14:
OK, so if I autolevel first, with the tracks code loaded, I then need to say no to saving but also no to deleting as I load each file, then they will all be auto levelled.  Is that correct?
What is the reason to allow saving?
Thanks
Mick

Harvie, 2019-04-15:
There are two use cases:

1.) You have flat g-code and load autolevel data (colorful grid with numbers) to bCNC, g-code is automaticaly autoleveled during streaming to GRBL. You can also store the autoleveling grid to .probe file.

2.) You want to make the autoleveling permanent and store it to g-code file. Then you have to apply to g-code, store the g-code. It is very important to remove probe data from bCNC, otherwise it will get autoleveled again when streaming. This makes sense in case you want to use the g-code in 3rd party software or machine that does not allow autoleveling itself.

For example if you are making PCBs and want to have perfect Z height precision, but every piece of material is different, then you keep the flat g-code and autolevel each piece separately as described in 1.)

Another example is that you want to use sharpie permanent marker to draw image on footballs. Precision is not extremely important, but you want to modify the flat image to fit on to the round football. So you can apply the autolevel data and produce g-code that you can send to your friend who wants to draw same image on his machine (maybe not even non-GRBL) and his football of the roughly same diameter. This is where you go for 2.) approach.

closed 2019-04-15 by Harvie

micksulley, 2019-04-15:
Thanks Harvie it is much clearer now.