vlachoudis/bCNC #1216: Unexpected move following alrm reset

micksulley, 2019-04-11:
Running 0.9.15 on a Raspberry Pi with an Ox
I have seen this a couple of times now, this time I was sure what happened.
I did a probe to set Z, jogged Z up a bit then jogged X & Y to my work zero, jogged Z down again to Z zero.
I then told it to probe and I got an error, which is expected as it is already sensing zero point.
I then pressed Reset, then Unlock.
At this point it set off at max speed in both +X and +Y, Z did not move.  Hit the emergency stop!
Why did it move at all?  I had not seen this on my previous version
Thanks
Mick

micksulley, 2019-04-11:
I have done more testing and this is reproducible. 
Probe with Z already triggered (not a normal occurrence but can happen)
Alarm(Z)  (as expected)
Reset
Unlock
Off goes the carriage in both X and Y.  My view is that is should need a positive action,e.g. start or resume, before anything moves at all.

Harvie, 2019-05-02:
Can this be related to #1208 ?

micksulley, 2019-05-02:
I have looked at #1208 and whilst it may be related I don't believe that it is the same thing.  The problem I see is that the machine moves on an alarm reset/unlock.  I haven't told it to do anything at all, just reset/unlock and off it goes.  As described above it is repeatable for me, does it behave this way for you?  I am happy to do further tests if there is any other information you need.

Harvie, 2019-05-06:
Do you use GRBL? Can you send us your $$ config?

micksulley, 2019-05-06:
Yes it is GRBL
$0=10
$1=255
$2=0
$3=6
$4=1
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=3
$24=25.000
$25=2500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=26.662
$101=26.641
$102=200.000
$110=2500.000
$111=2500.000
$112=750.000
$120=50.000
$121=50.000
$122=50.000
$130=820.000
$131=775.000
$132=68.000
Let me know if you need any more info or want me to try anything.
Mick

Harvie, 2019-05-06:
`$22=1` means you have homing cycle enabled...

MARIOBASZ, 2019-05-06:
I use homing cycle, but I also use soft limits $ 20 = 1, if I mistake the distance outside the limits, it avoids shocks. I have not suffered this problem.

MARIOBASZ, 2019-05-06:
Start cycle is very useful if I have to restart at the same point

micksulley, 2019-05-07:
When I enter $N I see -
$N
$N0=
$N1=
ok
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 To F0 S0]
ok

I tried setting $22=0  The problem is still there just the same, also the Home button does not work (obvious I suppose) so this is really not an option for me.
I also looked at the terminal output as the problem occurs, it shows
G53 G0 X0 Y0 Z0
I believe that G53 is absolute coordinates, so it is not driving to home but to absolute zero.

Why does it drive anywhere following a reset/unlock?  Surely you should have to give it an instruction before it moves at all?
If there is some reason to auto home it following a reset/unlock then 1) it should be to machine home not absolute zero and 2) it should home Z first to get the tool out of the way and only then home X and Y

Harvie, 2019-05-07:
> I tried setting $22=0 The problem is still there just the same.

Well so it's probably some other problem... Can you please launch bCNC from commandline and use `spy://` option in serial connection menu to capture serial communication?

micksulley, 2019-05-07:
Yes if you tell me how to use the spy:// option

Harvie, 2019-05-07:
[image]

micksulley, 2019-05-07:
I didn't realise that option existed :)
I have taken some screenshot which hopefully give the necessary data.
Screenshot2 is the point of the alarm, 3 is the reset/unlock 4 & 5 show the GRBL version (I think)

[image]
[image]
[image]
[image]

Harvie, 2019-05-07:
This seems to be the problem, but i don't know what causes it:

[image]

What do you think?

micksulley, 2019-05-07:
Yes looks like it.  How/why is that instruction generated?  This problem certainly didn't happen on the previous version that I ran.

micksulley, 2019-05-07:
Also I believe that this problem only occurs on reset/unlock following a probe Z already triggered alarm, not seen it after any other reset/unlock condition.

Harvie, 2019-05-07:
According to grep, there's only one line which can send such command:

```
./ProbePage.py:			cmd = "G53 G0 X%g Y%g Z%g\n"%(CNC.vars["prbx"], CNC.vars["prby"], CNC.vars["prbz"])
```

Maybe it gets buffered somewhere on bCNC side and the buffer is not cleared during reset, so it gets sent afterwards... We should clean that buffer during reset...

Harvie, 2019-05-29:
> That line is exactly the same in the version I was running before, did't have the problem with that

I know, but new version might buffer this line in different manner...

GitHubCaps, 2020-04-26:
@micksulley could this possibly be related to #1389 ? See the reply about "Automatic GOTO after probing" check-box. I realize this is an old post, but if others are having similar issues as this, there may be a need to handle this differently to achieve the desired results.

micksulley, 2020-04-27:
I have updated both issues, it does look to be very similar if not the 
same issue.

Mick