vlachoudis/bCNC #1217: Contact author please

CJELLWOOD, 2019-04-16:
Can I contact the author directly please regarding possible contract work

Harvie, 2019-04-16:
There are multiple authors. Depends on what part of bCNC you mean:
https://github.com/vlachoudis/bCNC/graphs/contributors

Is the contract work related to bCNC? Like adding new features to bCNC? I think it would be cool if somebody would be willing to pay developers for improving bCNC :-)

CJELLWOOD, 2019-04-16:
The situation is I am developing a GRBL controller board with my engineer who is attempting to include SDcard reader and LCD screen. He can do the hardware but says the firmware needs to be customised to work with the new features. He cant do that unfortunately. So I am looking for both a firmware developer to bring the features together and possible a trimmed down version of bCNC developed for quick plasma cutting.  Money is on the table and free boards of course to contributors

Harvie, 2019-05-02:
That might be possible. What kind of trimming are you talking about? Just removing unneeded features and buttons to make the UI bit more clean? That should be easy to do...