vlachoudis/bCNC #1219: Error while downloading nCNC via pip (pillow 6.0.0) -> Killed

MantaRayDeeJay, 2019-04-21:
I have an error while trying download bCNC via pip.
[image]
Hardware : Orange Pi Zero 512Mo with Armbian v5.75 Debian Stretch v4.19.20

sfinexer, 2019-04-22:
@MantaRayDeeJay   it's like so that did not work the command "clear" ????
This is not a single screenshot?
@MantaRayDeeJay  "python2 -m pip install --upgrade bCNC"

> sfinexer@sfinexer:~$ pip list
Package                            Version    
---------------------------------- -----------
alabaster                          0.7.8      
apt-xapian-index                   0.49       
asn1crypto                         0.24.0     
astroid                            1.6.5      
Babel                              2.6.0      
backports-abc                      0.5        
backports.functools-lru-cache      1.5        
backports.shutil-get-terminal-size 1.0.0      
bCNC                               0.9.14.57  
beautifulsoup4                     4.7.1      
bleach                             3.1.0      
cairocffi                          0.7.2      
CairoSVG                           1.0.20     
certifi                            2018.8.24  
cffi                               1.12.2     
chardet                            3.0.4      
cloudpickle                        0.8.0      
configobj                          5.0.6      
configparser                       3.5.0b2    
cryptography                       2.6.1      
decorator                          4.3.0      
defusedxml                         0.5.0      
docutils                           0.14       
entrypoints                        0.3        
enum34                             1.1.6      
funcsigs                           1.0.2      
functools32                        3.2.3.post2
future                             0.16.0     
futures                            3.2.0      
html5lib                           1.0.1      
idna                               2.6        
imagesize                          1.0.0      
ipaddress                          1.0.17     
ipykernel                          4.9.0      
ipython                            5.8.0      
ipython-genutils                   0.2.0      
isort                              4.3.4      
jedi                               0.13.2     
Jinja2                             2.10       
jsonschema                         2.6.0      
jupyter-client                     5.2.3      
jupyter-core                       4.4.0      
keyring                            17.1.1     
keyrings.alt                       3.1.1      
lazy-object-proxy                  1.3.1      
logilab-common                     1.4.2      
lxml                               4.3.2      
MarkupSafe                         1.1.0      
mccabe                             0.6.1      
mercurial                          4.8.2      
mistune                            0.8.4      
mock                               2.0.0      
nbconvert                          5.4.0      
nbformat                           4.4.0      
numpy                              1.16.2     
numpydoc                           0.7.0      
olefile                            0.46       
packaging                          19.0       
pandocfilters                      1.4.2      
parso                              0.3.1      
pathlib2                           2.3.3      
pbr                                4.2.0      
pexpect                            4.6.0      
[... 36 lines cut ...]
setuptools                         40.8.0     
simplegeneric                      0.8.1      
singledispatch                     3.4.0.3    
six                                1.12.0     
soupsieve                          1.8        
Sphinx                             1.8.4      
spyder                             3.3.3      
spyder-kernels                     0.2.4      
testpath                           0.4.2      
tornado                            5.1.1      
traitlets                          4.3.2      
typing                             3.6.6      
urllib3                            1.24.1     
vboxapi                            1.0        
wcwidth                            0.1.7      
webencodings                       0.5.1      
wheel                              0.32.3     
wrapt                              1.10.11    
wxPython                           3.0.2.0    
wxPython-common                    3.0.2.0    
sfinexer@sfinexer:~$ python -m pip install --upgrade bCNC
Requirement already up-to-date: bCNC in ./.local/lib/python2.7/site-packages (0.9.14.57)
Requirement already satisfied, skipping upgrade: Pillow>=4.0 in /usr/lib/python2.7/dist-packages (from bCNC) (5.4.1)
Requirement already satisfied, skipping upgrade: pyserial<=3.0.1 in ./.local/lib/python2.7/site-packages (from bCNC) (3.0.1)
Requirement already satisfied, skipping upgrade: numpy>=1.12 in /usr/lib/python2.7/dist-packages (from bCNC) (1.16.2)

sfinexer, 2019-04-22:
@MantaRayDeeJay But why do you have a new pillow 6.0.0 and absolutely all the other packages are old? Ie, your system is not evenly updated.

MantaRayDeeJay, 2019-04-22:
@sfinexer "Clear" cmd is OK and there is only one snapshot. I'm using Putty under Windows 10, so even after a clear cmd, I can scroll a bit !

MantaRayDeeJay, 2019-04-22:
@sfinexer I have tried the "python2 -m pip install --upgrade bCNC" cmd.
[image]

MantaRayDeeJay, 2019-04-22:
@sfinexer My linux distro seems to be OK.
[image]

sfinexer, 2019-04-22:
@MantaRayDeeJay I do not think that the error in bCNC, the problem you have.
For some reason, you do not download the package.
By the way, the pillow V6 package is not in the Debian repository.

MantaRayDeeJay, 2019-04-22:
> 
> 
> @MantaRayDeeJay But why do you have a new pillow 6.0.0 and absolutely all the other packages are old? Ie, your system is not evenly updated.

I don't know why, my system seems to be updated.

sfinexer, 2019-04-22:
https://packages.debian.org/search?suite=default&section=all&arch=any&searchon=names&keywords=python-pil

> stretch (stable) (python): Python Imaging Library (Pillow fork)
4.0.0-4: amd64 arm64 armel armhf i386 mips mips64el mipsel ppc64el s390x
buster (testing) (python): Python Imaging Library (Pillow fork)
5.4.1-2: amd64 arm64 armel armhf i386 mips mips64el mipsel ppc64el s390x
sid (unstable) (python): Python Imaging Library (Pillow fork)
5.4.1-2: alpha amd64 arm64 armel armhf hppa hurd-i386 i386 kfreebsd-amd64 kfreebsd-i386 m68k mips mips64el mipsel powerpcspe ppc64 ppc64el riscv64 s390x sh4 sparc64 x32
experimental (python): Python Imaging Library (Pillow fork)
6.0.0-1: amd64 arm64 armel armhf hppa hurd-i386 i386 kfreebsd-amd64 kfreebsd-i386 m68k mips mips64el mipsel powerpcspe ppc64 ppc64el riscv64 s390x sh4 sparc64 x32

MantaRayDeeJay, 2019-04-22:
@sfinexer I'm a bit newbie with Linux ARM devices, so I don't know where begin investigate, would you have some advices ?

sfinexer, 2019-04-22:
```
pip2 install --upgrade bCNC
pip2 install --upgrade git+https://github.com/vlachoudis/bCNC
pip2 install . #in git directory
python2 -m pip install --upgrade bCNC
```
I repeat that there is probably somewhere you have a problem. Try installing python-pil (pillow) yourself

sfinexer, 2019-04-22:
Just download the archive and that's it.

MantaRayDeeJay, 2019-04-22:
@sfinexer I have checked that I have well downloaded the last armbian stretch  version (v5.75) for my device (opi zero) but I will have to try with an older stable version (3.4 kernel) to see if it works better. Thanks for the tips.

sfinexer, 2019-04-22:
@MantaRayDeeJay Download the archive as shown by the link and run bCNC. You obviously have a bug in the peripheral software and not in the bCNC target program.

Harvie, 2019-04-22:
What happens if you try to install pillow separately?

`python2 -m pip install --upgrade 'Pillow>=4.0'`

Harvie, 2019-04-22:
Also check pip version by `python2 -m pip --version` and try upgrading `python2 -m pip install --upgrade pip`

Harvie, 2019-04-22:
If none of this helps, you should ask Pillow developers to help you with installing on the ARM (raspi):

https://github.com/python-pillow/Pillow
https://pypi.org/project/Pillow/2.2.1/

sfinexer, 2019-04-23:
@Harvie Debian and its package versions are independent of the processor architecture. With some exceptions.
In this case, somehow it happened that the versions became confused.
I assume that as a novice user could temporarily change the system update settings previously. Thus, some packages are updated from the experimental branch.
Back versions of packages, debian is not able to return, only reinstallation of packages, and then errors are possible.

Harvie, 2019-04-23:
> Debian and its package versions are independent of the processor architecture.

I am not talking about Debian packages. I am talking about PyPI. I've attached link to similar issue we had in past to my previous comment.

sfinexer, 2019-04-23:
I must admit that if there is a mistake, it is clearly not with us.
Sometimes, to prove an error in the third-party software is difficult, even leaving patches.
The main reason is the lack of a primary developer, ie the one who originally made the program. And those who now support software sometimes do not fully understand the specifics of the work.
This may apply to both PyPI and another program.

Many of the errors that were in my own software, it was possible to reveal after years! The end user does not understand the specifics of the program, may accept the error as a certain functional feature, and not report it.

MantaRayDeeJay, 2019-04-23:
@sfinexer I have clone the git projet and tried to launch bCNC. I have always some errors.
[image]

MantaRayDeeJay, 2019-04-23:
@Harvie pip2 are checked and up to date. I have also enough space on the SD card.
[image]

MantaRayDeeJay, 2019-04-23:
@Harvie Trying `python2 -m pip install --upgrade 'Pillow>=4.0'` without any success !
[image]
[image]

Harvie, 2019-04-23:
In the last screenshot you have the problem described. It needs zlib header files. Debian will probably call the package zlib-dev or something similar...

MantaRayDeeJay, 2019-04-23:
@Harvie I have checked and I can find a **zlib1g** package in the list, is it different from zlib-dev ?
[image]

MantaRayDeeJay, 2019-04-23:
@Harvie I have found this French tutoriel for Raspberry PI which using the Raspbian OS. This device is based on an ARM SoC as my Orange Pi Zero (Allwinner H2+). I know there are some differences between Raspbian and Armbian but these two distro are based on Debian for ARM architecture, so I think if it works well with the Raspberry, I would be able to use it with an Orange Pi !!!
French Title : **bCNC - Installation sur Raspberry Pi (Rasbian)**
https://www.youtube.com/watch?v=Mfoh6LVZaMA

sfinexer, 2019-04-24:
zlib1g,   zlib1g-dev  zlib1g:i386

MantaRayDeeJay, 2019-04-24:
@sfinexer zlib1g-dev installed. I don't think needed having zlib1g:i386 because I have an armhf  architecture.

MantaRayDeeJay, 2019-04-24:
@sfinexer The result below

```
pizero@orangepizero:~$ python2 -m pip install --upgrade 'Pillow>=4.0'
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Collecting Pillow>=4.0
  Using cached https://files.pythonhosted.org/packages/81/1a/6b2971adc1bca55b9a53ed1efa372acff7e8b9913982a396f3fa046efaf8/Pillow-6.0.0.tar.gz
Installing collected packages: Pillow
  Running setup.py install for Pillow ... error
    Complete output from command /usr/bin/python2 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-Jk9ZTI/Pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-Y4dqM_/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-armv7l-2.7
    creating build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/FtexImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/PpmImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImageFont.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/JpegImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/PsdImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/SgiImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/XVThumbImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/Image.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/_binary.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/PyAccess.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImageSequence.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/DdsImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/XpmImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/McIdasImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/IcnsImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImageTransform.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/BmpImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImagePalette.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/IptcImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/TiffImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/GdImageFile.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/_util.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/CurImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/TiffTags.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/FliImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/WebPImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImageGrab.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/WmfImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImageEnhance.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImageShow.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/PdfImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImageOps.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/GbrImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImageTk.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/BlpImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/PcfFontFile.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/WalImageFile.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/Jpeg2KImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImageMorph.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/TgaImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/EpsImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/_tkinter_finder.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/IcoImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImageWin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/GimpGradientFile.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImageDraw.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/features.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/Hdf5StubImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/JpegPresets.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/__init__.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/DcxImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImagePath.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImageDraw2.py -> build/lib.linux-armv7l-2.7/PIL
[... 221 lines cut ...]
    arm-linux-gnueabihf-gcc: error: build/temp.linux-armv7l-2.7/src/libImaging/RawEncode.o: No such file or directory
    arm-linux-gnueabihf-gcc: error: build/temp.linux-armv7l-2.7/src/libImaging/Storage.o: No such file or directory
    arm-linux-gnueabihf-gcc: error: build/temp.linux-armv7l-2.7/src/libImaging/SgiRleDecode.o: No such file or directory
    arm-linux-gnueabihf-gcc: error: build/temp.linux-armv7l-2.7/src/libImaging/SunRleDecode.o: No such file or directory
[... cut ...]
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

    ----------------------------------------
Command "/usr/bin/python2 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-Jk9ZTI/Pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-Y4dqM_/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-Jk9ZTI/Pillow/

```

MantaRayDeeJay, 2019-04-24:
@sfinexer I have opened an issue with Pillow. Thanks for your tips.
https://github.com/python-pillow/Pillow/issues/3806

sfinexer, 2019-04-24:
or zlib1g ?

MantaRayDeeJay, 2019-04-24:
@sfinexer zlib1g was already loaded in my linux distro by default, I think.

MantaRayDeeJay, 2019-04-24:
So now I have the 2 versions :

```
pizero@orangepizero:~$ sudo apt list --installed | grep -i zlib

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

zlib1g/stable,now 1:1.2.8.dfsg-5 armhf [installed]
zlib1g-dev/stable,now 1:1.2.8.dfsg-5 armhf [installed]

```

MantaRayDeeJay, 2019-04-24:
@sfinexer I have solved the pb. with Pillow : 
- I need 2 packages `sudo apt-get install python-dev python-setuptools`
https://pillow.readthedocs.io/en/latest/installation.html#building-on-linux
- I need to add the `--no-cache-dir` option for pip2 because sometimes the installation crashes cause of out of memory (RAM). I have only 512MB on my OPi. It explains why I have had sometimes the error message `killed` during the install process
`sudo pip2 install --no-cache-dir --upgrade Pillow`
- Do not forget using sudo to full success the package install.

MantaRayDeeJay, 2019-04-24:
@sfinexer Now I'm trying to install bCNC but the process seems to be crached during the "numpy" package. I'm connected via SSH and the process animation seems to be stopped as a "freeze" for more than 20mn.

MantaRayDeeJay, 2019-04-25:
I try to install all other packages manually before installing nCNC (pyserial and numpy)
pyserial is OK
[image]

MantaRayDeeJay, 2019-04-25:
Numpy installation is very very slow...

MantaRayDeeJay, 2019-04-25:
It seems, I have some problems to install numpy.

```
pizero@orangepizero:~$ sudo pip2 install --upgrade 'numpy>=1.12'
Collecting numpy>=1.12
  Downloading https://files.pythonhosted.org/packages/93/48/956b9dcdddfcedb1705839280e02cbfeb2861ed5d7f59241210530867d5b/numpy-1.16.3.zip (5.1MB)
    100% |████████████████████████████████| 5.1MB 34kB/s
Building wheels for collected packages: numpy
  Running setup.py bdist_wheel for numpy ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-YlBqhT/numpy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpxUXmTgpip-wheel- --python-tag cp27:
  Running from numpy source directory.
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help

  error: invalid command 'bdist_wheel'

  ----------------------------------------
  Failed building wheel for numpy
  Running setup.py clean for numpy
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-YlBqhT/numpy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" clean --all:
  Running from numpy source directory.

  `setup.py clean` is not supported, use one of the following instead:

    - `git clean -xdf` (cleans all files)
    - `git clean -Xdf` (cleans all versioned files, doesn't touch
                        files that aren't checked into the git repo)

  Add `--force` to your command to use it anyway if you must (unsupported).

  ----------------------------------------
  Failed cleaning build dir for numpy
Failed to build numpy
Installing collected packages: numpy
  Running setup.py install for numpy ... -

```

MantaRayDeeJay, 2019-04-25:
Oh yes !

```
Failed to build numpy
Installing collected packages: numpy
  Running setup.py install for numpy ... done
Successfully installed numpy-1.16.3
pizero@orangepizero:~$ sudo pip2 list
[sudo] password for pizero:
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
numpy (1.16.3)
Pillow (6.0.0)
pip (9.0.1)
pyserial (3.0.1)
setuptools (33.1.1)
pizero@orangepizero:~$
```

MantaRayDeeJay, 2019-04-25:
It seems that sometimes I mustn't use the "--no--cache-dir" pip option to install a package. A progress bar should be a useful help to avoid waiting a long time without to know if the process is crashed or not during a pip package install.
Work in progress... LOL

MantaRayDeeJay, 2019-04-25:
bCNC seems to be installed good !

```
pizero@orangepizero:~$ sudo pip2 install --upgrade bCNC
Collecting bCNC
  Using cached https://files.pythonhosted.org/packages/c8/cd/0e7f365c17cd101fe761b90f57a343fdceb4e3c4b7b9adb5eb05ce28359f/bCNC-0.9.14.57.tar.gz
Requirement already up-to-date: Pillow>=4.0 in /usr/local/lib/python2.7/dist-packages (from bCNC)
Requirement already up-to-date: numpy>=1.12 in /usr/local/lib/python2.7/dist-packages (from bCNC)
Requirement already up-to-date: pyserial<=3.0.1 in /usr/local/lib/python2.7/dist-packages (from bCNC)
Building wheels for collected packages: bCNC
  Running setup.py bdist_wheel for bCNC ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-IFlcT4/bCNC/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpEqzGyepip-wheel- --python-tag cp27:
  Running bCNC setup...
  /usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'long_description_content_type'
    warnings.warn(msg)
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help

  error: invalid command 'bdist_wheel'

  ----------------------------------------
  Failed building wheel for bCNC
  Running setup.py clean for bCNC
Failed to build bCNC
Installing collected packages: bCNC
  Running setup.py install for bCNC ... done
Successfully installed bCNC-0.9.14.57
pizero@orangepizero:~$ sudo pip2 list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
bCNC (0.9.14.57)
numpy (1.16.3)
Pillow (6.0.0)
pip (9.0.1)
pyserial (3.0.1)
setuptools (33.1.1)
pizero@orangepizero:~$
```

Next Step -> Testing it !!!

sfinexer, 2019-04-25:
@MantaRayDeeJay   You have all zapuskaetsya?
Could not be somewhat short form, if possible, to describe what you think was problematic.

MantaRayDeeJay, 2019-04-25:
After installed the "python-tk" package, I can launch bCNC on my OPi zero 512 !

[image]

MantaRayDeeJay, 2019-04-25:
@sfinexer Summary : 
To install bCNC on my OPi zero 512, I have done the following cmd :
**apt**
`sudo apt-get install python-pip python-setuptools python-dev python-tk libjpeg-dev zlib1g-dev`
**pip**
```
sudo pip2 install --no-cache-dir --upgrade 'Pillow>=4.0'
sudo pip2 install --upgrade 'pyserial<=3.0.1'
sudo pip2 install --upgrade 'numpy>=1.12'
sudo pip2 install --upgrade bCNC
```

Harvie, 2019-04-25:
I am really glad that you've managed to install this in the end.
I don't really know what to do about this.
Probably we should write some short installation manual, so other people can install it just like you did.
But still it seems to me that it's bit too complex. I wonder if there's easier way.
To be honest i've already lost track of all the stuff you had to do in order to install it. Can you please summarize it? Or maybe test it on fresh armbian install?

MantaRayDeeJay, 2019-04-25:
@Harvie I have tried to summarize it in my last message but I think it would be usefull to test it again on a new fresh armbian install as you suggest me !
Let me a bit of time to do it, I need to have a rest now ! it's tiring for a newbie in Linux environment !

Harvie, 2019-04-26:
@MantaRayDeeJay After you do the apt instal, in my opinion all of these commands:

```
sudo pip2 install --no-cache-dir --upgrade 'Pillow>=4.0'
sudo pip2 install --upgrade 'pyserial<=3.0.1'
sudo pip2 install --upgrade 'numpy>=1.12'
sudo pip2 install --upgrade bCNC
```

can be replaced by single:

```
sudo pip2 install --no-cache-dir --upgrade bCNC
```

MantaRayDeeJay, 2019-04-26:
@Harvie The problem is that pip seems not be very reliable with "low hardware setup". If I use the "--no-cache-dir" option, numpy installation "freeze". I will have to test it again later.

Harvie, 2019-04-30:
> pip seems not be very reliable with "low hardware setup"

Can you please figure this out with pip developers? https://github.com/pypa/pip

MantaRayDeeJay, 2019-05-02:
Please let me some time, I'm working on other issues with my devices. (OPi WiFi cnx and CNC device usb cnx) I will be back later, thanks.