vlachoudis/bCNC #1222: enlarge buttons

vair327, 2019-04-28:
Would it be possible to enlarge the buttons for each function i.e. zero axies and motion control, this would make much easier to use a touch screen.

Harvie, 2019-05-02:
Are you aware of this feature?

[image]

vair327, 2019-05-04:
This command does not increase the size of the control buttons only the icon for home and exit and such. I was asking about the buttons for setting zero and directional control they are small for touch screen.

Harvie, 2019-05-04:
[image]

closed 2019-05-04 by Harvie

vair327, 2019-05-04:
Perfect!

Thank You