vlachoudis/bCNC #1223: Slice mesh: MemoryError

miili, 2019-05-02:
Slice mesh runs into a memory error with an `.stl` 442 kBytes.

```
  File "/usr/local/lib/python2.7/dist-packages/bCNC/plugins/slicemesh.py", line 96, in execute
    verts, faces = self.loadMesh(file)
  File "/usr/local/lib/python2.7/dist-packages/bCNC/plugins/slicemesh.py", line 149, in loadMesh
    verts, faces = self.load_stl(file)
  File "/usr/local/lib/python2.7/dist-packages/bCNC/plugins/slicemesh.py", line 289, in load_stl
    verts, faces = self.merge_close_vertices(verts, faces)
  File "/usr/local/lib/python2.7/dist-packages/bCNC/plugins/slicemesh.py", line 244, in merge_close_vertices
    D = spdist.cdist(verts, verts)
  File "/usr/local/lib/python2.7/dist-packages/scipy/spatial/distance.py", line 2710, in cdist
    dm = np.empty((mA, mB), dtype=np.double)
MemoryError
```

Harvie, 2019-05-02:
Please share the .STL file otherwise i cannot help.

miili, 2019-05-02:
`.stl` can be downloaded from https://cults3d.com/en/3d-model/art/fairy-of-the-forest
bCNC screenshot, standard parameters from a fresh installation.

[image]

Harvie, 2019-05-02:
[fairy-of-the-forest20180624-11119-wpj9bg.zip](https://github.com/vlachoudis/bCNC/files/3138853/fairy-of-the-forest20180624-11119-wpj9bg.zip)

Harvie, 2019-05-02:
So i've managed to slice it. Took minutes to complete, but i was succesfull:

[image]

There is some problem with STL file being to big for STL algorithm. So i've used meshlab to open STL and exported it as PLY (standford polygon format, ASCII not binary!)

Most of the details lay around 3mm of Z height:

[fairy wood.ply.zip](https://github.com/vlachoudis/bCNC/files/3139019/fairy.wood.ply.zip)

Harvie, 2019-05-02:
I think we should fix this problem in future...  It's probably caused by numpy not being able to fit such big array/matrix... For now i might just add warning IDK...

Also please note that currently bCNC is not really able to do 3D CAM work... But you can get 2D contours and work with them as if they were 2D drawing (eg. DXF). Might be usefull for you or it might not...

miili, 2019-05-02:
Great, thanks for the prompt response!

MARIOBASZ, 2019-07-09:
I do not know if this comment is useful, I think a 3d route can be continued using 3d trochoidal