vlachoudis/bCNC #1224: AutoLevel Fast Probe Feed Issue

Skipper-is, 2019-05-04:
I am using autolevel, and have set the regular probe feed to 10, and the fast feed to 100, but the 100 for fast feed doesn't appear to be applied when moving between sample sites.
If I look at the Gcode:
```
G38.3Z-10F10
[PRB:51.803,0.055,-2.048:1]
ok
G0Z1
ok
G0X51.8028Y11.1778
ok
```
Surely on G0X51Y11 there should be F100?

MARIOBASZ, 2019-05-04:
G0 is fast or transfer speed (which you have set as maximum speed), does not carry F. G1 is milling speed, determined by F.
Read  Gcode no the web

Skipper-is, 2019-05-04:
And G0 should stay on the default feed rate or whatever it was initially set up on (the fast feed rate ideally), and not follow the probing rate, but yet it follows the feed rate of 10 from G38.3

Harvie, 2019-05-04:
Do you use GRBL?

Skipper-is, 2019-05-04:
The Smoothieboard

Harvie, 2019-05-04:
Duplicate of #1169

closed 2019-05-04 by Harvie