vlachoudis/bCNC #1225: Inverse z axis when auto leveling, opposite to gcode

Donsa90, 2019-05-10:
Hi, I searched through the guides and the issues for this problem, but cannot find a solution.

I generated gcode from eagle, and when used used in bCNC, the program runs good; The z axis will go up and then back down to carve the pcb. 

My problem is with the auto leveling; the z axis goes down and then up, as if it was somehow inverted (it tries to read the board in the wrong direction, y and x axis are in the right direction). I switched the wires for the z motor, and that makes it work.

The problem then is that the gcode and the auto leveling are in 2 different directions, so the board will be carved way too deep, because the readings from the auto level are opposite of what they should be. Do you guys have any idea of what might be causing the z axis to be inverted only during auto leveling?

Harvie, 2019-05-11:
Do you know that you are not supposed to apply autoleveling to g-code? it is applied automaticaly when sending to grbl. if you apply it, so g-code gets modified in editor, it means it will be autoleveled twice.

Donsa90, 2019-05-11:
so you mean that once the gcode is uploaded to bCNC, the autoleveling is automatically applied? I am not sure I understood. Also, even if I just do autoleveling (without g code), the same problem arises; the z axis is reversed, meaning that the bit goes down and then slowly up, trying to get contact with the board in the opposite direction.

Harvie, 2019-05-11:
Can you describe exactly what are you doing? Once there is autolevel data in bCNC (grid of numbers), it automaticaly autolevels. also note you have to set autolevel XY zero before proceeding.

Donsa90, 2019-05-11:
ok so first thing, I upload the gcode (at this point I can see the board and the machine position). After that, I make sure to set all axes equal to 0 where I want them. Then I go to probe->autolevel and click margins, zero and clear (at this point I see the grid and am able to set the number of squares). After all this is done, I press scan to start the autolevel, and here is where the problem starts: the z axis goes the opposite way.

Harvie, 2019-05-11:
You mean it starts probing upwards? Maybe change to negative number in Z here:

[image]

Donsa90, 2019-05-11:
Yes, I am in G54, and yes, it starts probing upwards. I changed to negative numbers like in the picture, but the problem is still there.

Harvie, 2019-05-11:
When you jog Z axis is it going downwards on negative and upwards on positive coordinates?
Have you set both MIN and MAX Z in autolevel settings?

What version of GRBL do you use? Have you considered flashing latest stable GRBL yourself? We've seen people with funky modified GRBL versions preflashed by board manufacturers. You are probably always better to flash GRBL yourself to be sure, there is no weird stuff happening.

Can you use `spy://` feature in serial connection menu to check g-code lines that are sent to your arduino?

Donsa90, 2019-05-11:
No, lets say z is at 0, if z goes to 1 then the drill bit is going down, if z goes to -1 then the bit is going up (is it supposed to go down on negative and up on positive? mine is the opposite). I have set both mix and max z, as shown in your picture (z mix=-3, z max=0, i also tried other combinations of positive and negative with the same problem). I already flashed the latest grbl on the arduino nano myself (v 1.1). For the spy://, where do I use that? I tried in the terminal window in bCNC (where it shows all the commands, it gives me an error there saying the command does not exist). Btw thanks for following up on the problem!

Harvie, 2019-05-12:
Z is supposed to go down on negative. If this is not a case in your machine, you can just turn wiring of your Z stepper motor around (switch the phases). If you use protoneer shield or similar, it's usualy enough to pull out the connector and plug it in other way round...

spy: is explained here: https://github.com/vlachoudis/bCNC/issues/1216#issuecomment-490018664

closed 2019-05-12 by Harvie

MARIOBASZ, 2019-05-12:
I'm not in my machine, but I usually put zmax 1 or some value greater than zero. I have not had this problem.
In that position, you will not be located in z less than -3 and therefore go up looking for your goal?

carl1961, 2019-05-12:
I have -5 for Z min  and 1 for Z max and works fine

Donsa90, 2019-05-14:
Yeah but turning the wire does not solve the problem. Although the z axis will go down doing that, and the autoleveling will work, the g code will be the opposite way now. So I will still have the auto level going down as it is supposed to, but the g code will go the opposite way. How to fix that?

Harvie, 2019-05-14:
@Donsa90 What? You should always cut to negative coords.

Donsa90, 2019-05-14:
So Im saying that switching the wire fixes the probing problem, making the probe go down (the right direction). The g code is the opposite direction though (that means that probing will go down, but g code is reversed)

Donsa90, 2019-05-14:
if you should alway cut to negative coordinates, how do you change that in the gcode? I checked on eagle and cant find an option to change to negative

Donsa90, 2019-05-14:
Actually my mistake in the explanation. The gcode already cuts in negative, but its the auto leveling that probes in positive. Switching the wires will make the gcode cut in positive and the the auto leveling go in negative (reverse)

Harvie, 2019-05-14:
Never done g-code using eagle, but usualy you specify cutting depth as negative number, when creating g-code.

carl1961, 2019-05-15:
I had this same issue yesterday, but it was just doing a probe. I forgot to zero my z axis, after I zeroed my z axis the probing went the right direction. I always probe my zero then I go do the auto level. I say your issue is because auto level did not have the new zero and it  was going to the last z axis zero. like me do you may have a z endstop (home switch) on your z max.