vlachoudis/bCNC #1226: How to set $xx not in the controller list

alexpGH, 2019-05-14:
HI,
I try to use bCNC for my mpcnc with dual axis auto home.
Therefore I need to set e.g.$104 and $114 (x and y axis are cloned to grbl-mega-5x 's A and B axis). People use the grbl-mega-5x code for this kind of machine with UGS, which works well.
In the attached config file for UGS there are a set of $xx as mentioned above which I don't know how to set in bCNC. Would it be possible?

Harvie, 2019-05-14:
It's not currently supported in UI, however you can just switch to bCNC terminal window (button in top-right corner) and enter the commands manualy. Eg.: `$104=something` and `$$` to verify it was stored.

closed 2019-05-14 by Harvie

alexpGH, 2019-05-14:
ah, cool!
Thanks!