vlachoudis/bCNC #1227: Tool change

Luftlupus, 2019-05-18:
Any advice would be greatly appreciated.
I am trying to do a tool change on my CNC but after it probes it is giving me an error saying its trying to move 30mm on the Z and its outside of machine limits? should it not just go back over where the code left off then go back down to what it was doing? I am obviously missing something here. Cheers
i know the limits are fine as i have been using the CNC fine without tool change being implemented. link to the video below. 

I have tried looking everywhere but i am at a loss and still pretty new to this..

https://www.youtube.com/watch?v=Mm--M4C92Ns

DeanRM, 2019-05-20:
I'm going through some growing pains myself.

1)  Tool numbers are not used by grbl or bcnc.  They are for automated tool changers.  In your vcarvepro postprocessor, in the tool change area, remove any reporting for T<n>.  (T01, T02, etc...)

2)  Mount your touchplate on your spoil board.  I have mine mounted bottom left.  Attach the plate to your grbl probe pin and your clip to probe ground.  
[image]

3) Under the probe tab and tool icon,  you have some data to set - Tool Change location and Tool probe location.  You do this by moving your router to the desired spot and then press the get button.  For the probe location - you move the router centered above the probe high enough to hold your longest bit without touching the plate.  My longest bit is 4", so I'm at Z-10.  
Select the type of tool change you want - WCS or TLO.  With a fixed touchplate - use TLO  otherwise you need WCS.
Next change your probe travel distance - since my longest bit is 4", I set my travel distance to 77mm.
Write all this info down on paper for the next step.

4)  To keep this information after a reboot - you have to update bCNC's user file.  In the tool change area, add the change + probe location, change type (WCS = 3, TLO  = 4) and distance.  Save the file and restart bCNC.  Check the probe tab and tool icon, your data should still be there.

Your touchplate is now setup.
To use it - zero your material, calibrate the first bit under probe tab/tool icon (remember to attach the clip), then run your gcode.

Tip - grbl $10 = 3 for proper reporting.

Luftlupus, 2019-05-20:
This is a great write up and i have got this working now but i have run into an issue in the tool change after i swap to the next bit and it probes works and all but runs into an alarm after first tool change just when it has goes to go back to work position due to a machine limitation? i can only assume its trying to move +up 5mm before it goes back to the job but i set the tool change position 5mm below the home switch on the Z. also tried setting it another 5mm down but same issue.
[image]

DeanRM, 2019-05-20:
Your Z travel distance is too short.  Most bits are two inches so your Z probe position is to high.  My max Z travel is 90mm, that is why my position of Z-10 and a travel distance of 77mm.   Your settings will be different.  Please redo your setup for your machine.

Luftlupus, 2019-05-22:
thanks for the help issue was resolved.

closed 2019-05-22 by Luftlupus

Harvie, 2019-05-22:
@Luftlupus Can you please give us some brief description of what problems you have faced, how did you solved it and if there's something we can improve in wiki documentation to be more clear for newcomers?

DeanRM, 2019-05-23:
Probe setup for a tool change needs better documentation.   Nothing is mentioned about what to do for a fixed location verses using a probe on your workpiece.  What WCS vs TLO choices mean and when you should use either one.  Also, nothing is mentioned about persisting a fixed probe location so a reboot does not lose the change/probe location.  Had to figure that out on my own.   

I'm also not happy about tool change and how it works.  I've had errors during change/calibrate that required a hard reset to break out of and had to start over on zeroing z.

carl1961, 2019-05-23:
@DeanRM, have you read this? https://github.com/vlachoudis/bCNC/wiki/Tutorials:-Tool-Change-with-bCNC,-FreeCAD-&-grbl

Luftlupus, 2019-05-23:
The first problem was the code from Vcarve it was not setup correctly for tool changes in Gcode i had someone share there Tool change profile for the software so that fixed that.

Next problem was after calibration of the probe which was calibrated in the negatives instead it should have been a positive number in the calibration field which i just changed then that was fine.

last problem was when just after probing a tool change would get an error saying it was trying to +Z 5mm but because Vcarve had the safe z set to 5mm up it was pretty much trying to go up to the limit switch because i had the tool change position set 5mm below z max height, so i set Vcarve safe z to be 3mm above the work piece and set the tool change position 10mm 5mm below z limit meaning it could travel 3mm up z after a tool change and have 2mm height spare stopping that error.

So i would say wiki could add a little more info but it all comes down to what design programs and code generation you are using in the end its hard to have info for everything the only reason i had this fixed was someone on another forum was very helpful and i cant thank them enough for helping me sort it out.
The community around all this is awesome and i am glad to be apart of it. =)