vlachoudis/bCNC #1228: Annoying bug with combobox

DeanRM, 2019-05-20:
I am running bcnc on a raspberry pi and I have an issue whenever I change the distance for the axis.  See picture - the combobox is not populated.   I have to scroll a bit to get it to populate.
[image]

Harvie, 2019-05-22:
Weird. I don't have this problem... Can you try deleting your bCNC.ini?
Also what version of python 2.7 do you run?

carl1961, 2019-05-22:
Increase your timeout time
[image]

DeanRM, 2019-05-23:
Not sure,  but I am using the canned setup from Protoneer so it is very outdated.  Running updates in everything.  You can close this issue.  Sorry to bother you.

Harvie, 2019-05-24:
So have you tried upgrading bCNC?

DeanRM, 2019-09-04:
I installed a fresh sd card and installed the latest everything.  Good learning experience.  And yes, I still get the blank combobox.  Anyway to add refresh to the box after opening?