vlachoudis/bCNC #1229: bCNC not connecting to Gecko G540

leftbrainstuff, 2019-05-22:
I'm running bCNC on Ubuntu Linux with a parallel port connection to a Gecko G540. The G540 is functioning and holding motor torque. bCNC is also running, connecting to /dev/ttyS0 and the Dell Optiplex is set to allow /dev/ttyS0 to operate in parallel mode.

The bCNC status remains orange but says connected to grbl when clicked. I can also connect to the G540 using the java app Universal Gcode Sender but it pops up a message indicating grbl is waiting to boot. This appears to be a connection issue and not a bCNC or G540 issue AFAIK.

I've seen some mention of $10=3 (an odd value) and position (POS) settings needing to be customized in bCNC. However there is nothing in the bCNC documentation and a search of the issues yields no similar issues. I've also reached out to Gecko support to see what tests / checks I can conduct to root cause this issue. Their documentation is also quite lightweight in terms of troubleshooting connectivity. I'm awaiting for a response.

What tests / checks can I do to troubleshoot the connection issues between the G540 and the db25 plug on the host machine? I've used a number of linux tools (setserial, stty, minicom, etc) but am not seeing any transmission through the db25 connector.

LYHTSPD, 2019-05-22:
One thing I noticed with the $10 parameter is I needed to enter the number into the field and then press the enter key.  All of the other values seem to update live, but that one did not.  Maybe give that a try?

Harvie, 2019-05-22:
I don't understand what your setup is, but as far as i know, parallel connection is not supported by GRBL nor by bCNC... Only serial.

leftbrainstuff, 2019-06-16:
System is bCNC running on Ubuntu Linux laptop, ethernet connection to Mesa 7i92 board flashed to G540X2D. Then the mesa 7i92 board connects to the G540 using a parallel port. I can ping the board, verify the FPGA is flashed for the G540 but cannot generate output from the mesa card.

Harvie, 2019-06-19:
Does the Mesa 7i92 have GRBL or SMOOTHIE compatible interface? If you want to use bCNC with such machine, you would probably need to use arduino with GRBL connected to the parallel stepper driver. This can be as easy as flashing GRBL to arduino and running bunch of wires from arduino to your parallel board. You might also find ready made adapter. Or just ditch the parallel board altogether and use protoneer shield instead. It's really cheap on aliexpress...

closed 2019-06-19 by Harvie

leftbrainstuff, 2019-07-01:
Thanks I'll check that out.