vlachoudis/bCNC #1233: Autolevel values inverted when position X = 0

janschiefer, 2019-05-28:
I have been using the current bCNC development version with good success for some weeks now. I'm using it with an Chinese CNC 3018 supporting GRBL 1.1.

However I have noticed a bug with autoleveling:

The autolevel data on the X = 0-marigin always has the wrong decimal sign (see attached image). For example -0.3250 on the origin position should be 0.3250. -0.2930 just under it should be 0.2930.

If I set smaller steps this is reproducible. The next value where X
[image]
 is bigger than 0 is probed correct, while all values with x=0 are inverted.

As a workaround I have to probe an area, that is actually a bit bigger than my PCB...

Harvie, 2019-05-29:
Have you seen #1225 ?

janschiefer, 2019-05-30:
My coordinates are set in a downward way.
It just happens for X = 0. For any other X > 0 the autolevel data is correct.

I don't think issue #1225 is related...

MARIOBASZ, 2019-05-30:
I have never had that inconvenience. Maybe you should update the autolevel grid by raising or lowering the number of steps X and Y?

janschiefer, 2019-05-30:
I can change the autolevel step number to a very small number to mitigate this, so only a few millimeters on the left of my board are affected. However if I have a normal step count autoleveling a larger area of my board on the left side is affected by incorrect depth cuts.

Harvie, 2019-06-29:
> My coordinates are set in a downward way.

What does it mean?

> I can change the autolevel step number to a very small number to mitigate this, so only a few millimeters on the left of my board are affected.

This is weird. Especialy the image you've attached in the beginning. Unless there's some physical step in your setup... We should figure out what causes this. But it does not happen to me, so it's kinda hard to debug. Do you have at least some basic knowledge of python?

janschiefer, 2019-06-30:
Im actually on vacation now without computer. We can try to debug this in
the next weeks.