vlachoudis/bCNC #1234: Crash at GUI startup since commit 33da2ad

ted0xff, 2019-05-31:
I was using version d6f90a490416e85b9dab9065d070a82d5af94c11 of 30 dec 2018 with no problem. I tried to update to have latest features but the GUI crash at start-up with version f71b790.

Traceback:
```
$ ./bCNC.sh 
Launching bCNC from .
_GenericGRBL
Traceback (most recent call last):
  File "/home/teddy/dev/CNC/Appli/3_gcode_send/bCNC/bCNC-git/Sender.py", line 275, in controllerLoad
  File "<string>", line 1, in <module>
AttributeError: 'module' object has no attribute 'Controller'
GRBL0
GRBL1
_GenericController
Traceback (most recent call last):
  File "/home/teddy/dev/CNC/Appli/3_gcode_send/bCNC/bCNC-git/Sender.py", line 275, in controllerLoad
  File "<string>", line 1, in <module>
AttributeError: 'module' object has no attribute 'Controller'
SMOOTHIE
test supergen grbl
test supergen grbl
ctrlist
test supergen grbl
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/teddy/dev/CNC/Appli/3_gcode_send/bCNC/bCNC-git/bCNC/__main__.py", line 2678, in <module>
    main()
  File "/home/teddy/dev/CNC/Appli/3_gcode_send/bCNC/bCNC-git/bCNC/__main__.py", line 2646, in main
    application = Application(tk)
  File "/home/teddy/dev/CNC/Appli/3_gcode_send/bCNC/bCNC-git/bCNC/__main__.py", line 244, in __init__
    self.ribbon.addPage(self.pages[name],side)
KeyError: 'CAM'
bCNC-git $
```

I did the bissect and this is commit 33da2ad, related to CAM, which introduce the problem.

I think I need to install a module/dependencie, but I don't know which one...

I don't use any camera feature (yet) and there is no camera plugged on my computer.

I revert to dd4b603, commit just before the problem, my CNC works well with this one

Harvie, 2019-05-31:
Please read documentation on how to use pip to install bCNC

ted0xff, 2019-05-31:
OK, my bad, reinstalling from pip2 instead of using a git checkout works better.

Thank you very much for your reactivity and your work !

closed 2019-05-31 by ted0xff