vlachoudis/bCNC #1236: camera not working bcnc raspberry pi

mikeymp, 2019-06-02:
My bcnc install works well on my raspberry pi 3b i just need to know where in bcnc do you set up a usb camera? All packages needed are up to date and newest as of today. Cannot get crosshairs or video screen for bcnc to show but icon is no longer grayed out just the camera setting tool .Where in bcnc do you make the changes to get camera to work?

mikeymp, 2019-06-08:
i now have a good install of basic opencv detects my camera i ran calibration .py video.py and videov412.py from opencv folder camera works on all but not detected by bcnc cv2 also returns true no errors.

mikeymp, 2019-06-09:
it all works good now camera is up and working and works well even over ssh .I was the problem all software works with no errors. Thank  you for providing a great product that has awesome features and somewhat easy to use for a new guys first Cnc build.

Harvie, 2019-06-10:
duplicate of #1235

closed 2019-06-10 by Harvie