vlachoudis/bCNC #1237: HOW instal bCNC in Linux MINT 19.1 MATE 64 bit (Ubuntu 18.04.1)???

lumanga, 2019-06-04:
Dear Sir
I try to install bCNC on my MINT 19.1 MATE 64bit, I followed the guide:
`
sudo apt-get update
sudo apt-get upgrade
sudo apt-get install git python python-tk python-pip
`
in but I'm receiving thir error when i do:

linux@Mint19:~$ **sudo pip2 install --upgrade bCNC**
`
Collecting bCNC
  Downloading https://files.pythonhosted.org/packages/c8/cd/0e7f365c17cd101fe761b90f57a343fdceb4e3c4b7b9adb5eb05ce28359f/bCNC-0.9.14.57.tar.gz (1.2MB)
    100% |████████████████████████████████| 1.2MB 694kB/s 
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    ImportError: No module named setuptools
   ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-zWSQL4/bCNC/
`

Harvie, 2019-06-05:
Can you install python module named setuptools? Never had this problem, i think setuptools is expected to be part of python or pip i always had it installed by default...

```
sudo pip2 install --upgrade pip
sudo pip2 install --upgrade setuptools
```

lumanga, 2019-06-06:
SOLVED.
doing this:

sudo apt-get update
sudo apt-get upgrade
sudo apt-get install git python python-tk python-pip
sudo apt-get install python-setuptools
sudo pip2 install --upgrade bCNC

closed 2019-06-06 by lumanga