vlachoudis/bCNC #1240: Memory access fault on Raspi

poesel, 2019-06-10:
(Debian Stretch, updated to current on Raspi 2)

After a longer hiatus I wanted to start up bCNC again. It didn't work so I updated with
'pip2 install --upgrade bCNC' as root.
This worked (after i had 'libjpeg-dev' installed).

Then as user:
markus@fraese:~$ python2 -m bCNC
new-config Utils <ConfigParser instance at 0x75e97e90>
Speicherzugriffsfehler
markus@fraese:~$ 

 Speicherzugriffsfehler = memory access fault

? what did I do wrong?

Thanks

Harvie, 2019-06-10:
Can you please upgrade your python?

poesel, 2019-06-10:
I have this version of python2:
python2.7/stable,now 2.7.13-2+deb9u3 armhf  [Installiert,automatisch]
  Interactive high-level object-oriented language (version 2.7)

I've uninstalled pillow (with pip). Open-cv is not installed. 

Still the same fault. What can I do to get a more informative fault message?

poesel, 2019-06-12:
Sorry for the trouble - the cause was a failed upgrade, nothing to do with bCNC. The upgrade ran without problems but the system wasn't updated for a long time so I had to dist-upgrade.
After that bCNC ran without problems.

closed 2019-06-12 by poesel