vlachoudis/bCNC #1241: bCNC not connected to CNC but Connection is establish with Grbl

excury, 2019-06-11:
Hi I am unable to connect my controller to bCNC. I open the serial port but the status stay on Not connected with the orange background. It also said Connection is establish with Grbl, see screen shot.
> 
> [image]

The control module does not work, I cannot control the motor. When I send a command, noting happen.
Can anyone help?

I'm using a Rasberry Pi.
Ive tried turning on and off the serial port.
I have tried the $10=1 that was propose earlier.
I have tried to do a software reset  of the controller and the hard reset but it does not seem to complete.
V 0.9.14 8th Jan.
 
I am using a usb to connect to Eleksmaker Mana Briver Board.
I am using TTYusb0

If anyone could help me to solve this issue that would be awesome!

tarbard, 2019-06-13:
I had the same problem until I set $10=3, now it's fine again 

I'm on GRBL 0.9, The docs seem to suggest $10=1 as the safe default for this version of GRBL but I get the same connection problem as you do when set to that.

excury, 2019-06-14:
I change the set $10=3 but the status did not change. 
The motor knob are now hard, that is a change, so I suppose there is communication. But I cannot control the motor from the control panel. When I press on the arrow to move the laser, nothing happen.

Harvie, 2019-06-19:
Please try latest pip version. Also maybe upgrade python to latest 2.7

BlakusSenior, 2019-06-20:
It's not connected but Ihave bCNC installed and working but I cannot call up the configuration menu ie. entereing $$ does nothing. Anyone help please?

Harvie, 2019-06-29:
@BlakusSenior please open separate issue.

closed 2019-07-15 by excury

Harvie, 2019-07-16:
@excury have you resolved the problem?

BlakusSenior, 2019-07-16:
Yes thanks Harvie all good.

BlakusSenior, 2019-07-16:
Whoops !!

Harvie, 2019-07-16:
> Yes thanks Harvie all good.

And bCNC still says "Not connected"? Can you please use bCNC spy:// feateure to capture serial communication to see what's going on?

excury, 2019-07-16:
Yes, thank for asking,  the micro-USB connector on the controller was broken. I fix it and now it work.

Harvie, 2019-07-16:
LOL

kd6vpe, 2020-12-07:
I just installed pip install bCNC on a rpi 3+. I can not get it to show connected status. It runs g-code files but does not show updates to bit location on the grids nor the files image. Anyone have any insite to this problem. It used to work on my old pi2.

AlvaroRey, 2022-01-24:
Comment from 2022 with the same topic, This problems happens to me from time to time, what I do is to Upload an Epprom clear code using the arduino IDE to my Arduino UNO ( in my case, I'm using a CNC shield)) and then reinstall GRBL. Usually this works for me. Just in case, someone is dealing with this problem, and modify de $10 doesn't help at all.

Carmah565, 2022-06-30:
i have this problem for a couple of days now, i reinstalled bcnc, python ,pip , arduino grbl, i tried to change the status 10$, but bcnc does not accept any command in the terminal and continues as not connected. Does anyone have any other suggestions?

luckypickle1, 2024-06-24:
I already had this problem, I solved it by replacing the Arduino IC chip with another one.

Carmah565, 2024-06-25:
I solved my problem at the time, I was using a valve to activate a cylinder in my machine, and this valve was creating a counter electromotive current and affecting the USB input communication. I put a diode in the valve and it solved it easily, but it took me a while to figure it out.