vlachoudis/bCNC #1243: Controlling RPi GPIOs with python scripts in the GCode possible?

ill-phil, 2019-06-25:
Hello Guys, 
first of all thanks for this great software! I'm using it on all my cnc machines, running it on a RPI and so far not missing any feature. Now i'm planning to build a more complex machine. Therefore, i have to use more I/O-signals. I'm thinking about using python scripts (%...) inside the GCode to interact with the RPi GPIOs. 
So, my question is, what do the experts think about this approach? Has anyone tried this already? Are there any limitations to the python scripting function in bcnc. Latency will not be a problem in my case, since i don't have time critical I/Os.

To show you, what i will use for, check out my website: www.rootpecker.com

Harvie, 2019-06-25:
Yes. This is possible. However i am not sure about timing. Because g-code goes to GRBL buffer, while python commands are executed localy. This means there might be some g-code commands unprocessed by GRBL at the time your python commands are executed. Luckily there's `%wait` command to wait for it, before you execute the python.

https://github.com/vlachoudis/bCNC/wiki/Scripting

closed 2019-06-25 by Harvie

MARIOBASZ, 2019-06-25:
hello @ill-phil,  congratulations for what you do.
Curiosity, what do you try to add?
If you do it with python, I would like to see it, I have few python knowledge.
I am adding things to my machine, (vacuum table, ATC,), I modified in grbl files creating new M commands that accept the machine, I enabled new bolts.
regards

ill-phil, 2019-06-25:
Hi, 
thanks for your answers and the tip with `%wait`. That encourages me to try it that way. I will add: a vacuum table, a 4th axis (which doesn't need to be synchronized), a temperature control, some pneumatic actuators and some more endstop-like switches. As soon as i have made any progress on that, i can share it here if you are interested in. 
Regards Philipp

MARIOBASZ, 2019-06-26:
I'm going to read how you implement it, I'm also going to need more than one rotary axis (one of them to mill double-sided PCB), you can also see that there is a version of Grbl for 6 axes, although I do not know if it really works.
I have implemented the activation of a motor in both directions as I have said, adding pin and new number of M in the files of grbl

MARIOBASZ, 2019-06-26:
I am also doing tests with pneumatic actuators in my machine, in what is my objective of a cnc with different or uncommon features, I am confident in being able to implement my ideas in reality

LYHTSPD, 2020-03-27:
Were either of you able to figure out how to access the GPIO pins through script?  I am wanting to wire up a stack light and indicate program running, tool change needed, etc.  Maybe even wire in some other sensors.

flemish4, 2020-04-01:
I have GPIO working through scripting. 
I got the GPIO doing what I wanted through an external python script running outside bCNC. Then I used:
%import this_is_your_python_module
%this_is_your_python_module.run_function()

Which all works fine although I don't seem to be able to integrate it into a gCode file.

MARIOBASZ, 2020-04-02:
I'm interested in the progress you make. I have it on Arduino.
My way was to declare new M code, make the changes and declare the pins in cpu map and the rest of the changes. If there is another way I am interested in reading it.