vlachoudis/bCNC #1244: Can't get setup list on bCNC

BlakusSenior, 2019-06-30:
I have bCNC installed and working but I cannot call up the configuration menu ie. entereing $$ does nothing. Anyone help please?

Harvie, 2019-06-30:
Do you have latest pip version of bCNC, latest python 2.7 and latest GRBL installed on your arduino? You should flash GRBL yourself to the arduino. Never trust any 3rd party vendor, they tend to flash strange, buggy or even fake GRBL versions to their boards.

BlakusSenior, 2019-07-07:
I have python 3.7.2 and the latest bCNC and have installed/flashed  grbl to the Arduino board a couple of times using the sketch from the examples section in the files menu of arduino ide, everything works when bCNC is up ie. jogging etc but I cannot access the setup menu with the $$ command.

Harvie, 2019-07-08:
Have you flashed latest GRBL 1.1 version? https://github.com/gnea/grbl/

Have you switched to terminal tab in bCNC?

[image]

BlakusSenior, 2019-07-09:
I'll check on those items

BlakusSenior, 2019-07-13:
Thanks Harvie,
I had not connected as terminal mode.
So I can go back to messing up other stuff :-)

Harvie, 2019-07-13:
Glad to hear it works for you.
Problem is that when you are not in terminal mode, you can send the commands, but you will not see the GRBL output, so you can list grbl settings, but you will only see it in terminal.

closed 2019-07-13 by Harvie