vlachoudis/bCNC #1245: AttributeError: DROFrame instance has no attribute 'wcsSet' using Set WPOS

frado65, 2019-07-01:
Hi, I've updated the bCNC (0.9.14-dev) on my ubuntu 18.04 but now Set WPOS seems do not work, if I click it bCNC freeze and the only solution is to kill it.
The console write message "AttributeError: DROFrame instance has no attribute 'wcsSet'".

Harvie, 2019-07-01:
Just checked and i can confirm this. We need to fix this.

Harvie, 2019-07-01:
Can you check agian pls?

closed 2019-07-01 by Harvie

frado65, 2019-07-01:
Hi, I've added a screenshot about the new test.
[image]

frado65, 2019-07-01:
Now I've applied the patch and it seem to work! :) :+1: 
[image]

commit 14c0c776 (2019-07-01): Fixed setting DRO using canvas cursor. Fixes #1245