vlachoudis/bCNC #1246: CNC 2417 MILLING Z AXIS AND PROBE AND COMMAND PROBLEM

zaki554, 2019-07-03:
we want to know that how the z axis will be set and how we get the thickness by installing the probe and please tell about the command how should it be used in GRBL CONTROL 0.8

Harvie, 2019-07-03:
1.) Please upgrade your firmware to latest GRBL (currently version 1.1f). Version 0.8 have some issues. You can download it here https://github.com/gnea/grbl also they have instructions on how to install.

2.) Some instructions on autoleveling are here:

https://github.com/vlachoudis/bCNC/wiki/AutoLevel
https://www.youtube.com/results?search_query=bcnc+autolevel

closed 2019-07-03 by Harvie

zaki554, 2019-07-04:
Thanks..kindly tell me how can we set the depth and find the distance by
probe and what is hard limit error.I am using the B CNC software.

Harvie, 2019-07-08:
Do you have hardware ready for this? Probe correctly connected to arduino with grbl?

These are relevant controls in bCNC to start experimenting:

[image]