vlachoudis/bCNC #1247: Invert Y axis

Dan018, 2019-07-03:
My CNC machine homes to the top left corner of the machine, with +X moving right, and +Y moving down (coming towards the user). Homing works fine and HOMING_FORCE_SET_ORIGIN is enabled so MPos is 0,0,0 when at the top left corner after homing.

bCNC GUI has X moving right, but the Y axis moving up. Is there a setting to change this so the Y axis is down rather than up?

Harvie, 2019-07-03:
Is there reason why your machine is set like this? (it's very easy to change this direction by swapping wires to Y stepper motor(s)). It's pretty standart to have machines set up like this:

[image]

If you REALLY can't fix this, you can also swap the direction in GRBL settings, but i would reccomend you to fix the wiring instead.

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#3--direction-port-invert-mask

closed 2019-07-03 by Harvie

Dan018, 2019-07-03:
I've set the machine up like this so that the X gantry is at the opposite end of the machine to the user so when I'm putting material on to the spoil board, or removing cut parts, I don't have to reach over the gantry.

I could set up a macro to go to the top left, load workpiece and then home in the bottom left if there's no way in bCNC to change the Y axis.

Harvie, 2019-07-03:
> I've set the machine up like this so that the X gantry is at the opposite end of the machine to the user so when I'm putting material on to the spoil board, or removing cut parts, I don't have to reach over the gantry.

I don't understand what you mean. This is not related to axis settings at all. You can have gantry (or origin/zero) in any place you need, no matter if you invert Y or not.

Dan018, 2019-07-03:
No it's more a GUI thing. The motion works fine in that all axis move as they should, except that my Y axis moves from the back of the machine to the front, which is what I want it to do.

The GUI starts in the bottom left (not top left) so when I move the Y axis in a positive direction the GUI moves up the screen, while the gantry moves (correctly) down.

Essentially I wanted to make the +Y axis move down in the GUI to be more representative of the work area, but I think this may not be possible.

Harvie, 2019-07-03:
> Essentially I wanted to make the +Y axis move down in the GUI to be more representative of the work area, but I think this may not be possible.

Even if you would made this possible in bCNC, you will still get mirrored part when compared to your original CAD/CAM inputs...

I am still not sure what are you trying to achieve, but i am almost sure it can be done without swaping axis direction... Eg.: Do you know you can do homing in any direction you want without swaping the coordinate system around???

Dan018, 2019-07-03:
Yes, I've used $23 to get the homing sequence correct.

The top left of my machine is 0,400. If I could have the machine run a homing sequence and then know its position is 0,400 and not 0,0 that would work. I'll see if I can achieve that. Thanks for your help.

Harvie, 2019-07-03:
Well. When you home, the top left will be 0,0 in **MCS**. You might then set **WCS** for it to be (0,400).
Also i would ask GRBL developers what is preffered way of setting nonzero coordinates in homing point.