vlachoudis/bCNC #1248: Error 33 when using editor

Paulmsam, 2019-07-03:
Hi :)

First off, thank you for the fantastic job on this software! Been getting great results, it's very much appreciated :)

What has happened, I created a design in fusion360, and exported it via the grbl post processor built in. 

What I had done is not oriented the X or Y in the correct direction, to fix this I used the rotate command. And then would set the origin to where I need. What would happen is I would get a error 33 at the first arc usally, this happened a few times, more than what I can tell is normal online, as it seems fusion has an issue with arcs. What I had noticed, if I sent it out of fusion in the correct orientation, and did not do any rotation or change the origin, the same path would not give me the 33 error. But as soon as I rotated it it would error out. I have to cheack each g code with the $c command, which catches the issue. 

I am using the most updated version off all software, and firmware.

Harvie, 2019-07-03:
Honestly i don't know what is wrong. Do you care to figure out the error in g-code yourself?
I know people had some issues with g-code from fusion in bCNC, you might find some details here:
https://github.com/vlachoudis/bCNC/issues?utf8=%E2%9C%93&q=fusion

However if you want someone to help with the detective work, i can suggest you to upload sample of-gcode here. Prefferably some example code for milling small part, so the g-code file is short and therefore easy to examine how the rotation goes wrong.

Paulmsam, 2019-07-03:
Hi Harvie :)

I will try recreate the issue and upload the g code asap :)  I have unfortunately overwritten all the one ones that gave the issue.

michael-keymont-bose, 2019-10-23:
FWIW, I was playing around with the tool change function yesterday and, needing a quick and simple "job" to run between tool changes, I wrote up a very basic g code program that simply created a circle using 2 arcs. I also see the Error 33. No Fusion 360 involved at all, so it MAY be an issue with arcs, specifically. There was virtually nothing in that g code besides the 2 arcs and a couple of z height adjustments...

MARIOBASZ, 2019-10-23:
can you copy the code? I don't remember if error 33 is due to an error in the I J values ​​to generate arc

michael-keymont-bose, 2019-10-24:
Hi,
I actually was just going through a GCode tutorial, so I’m pretty sure the code is good, but here it is:

G00 X0. Y0. .75
G00 X1.5 Y1.
G02 X1.5 Y2. I0. J.5 F30.
G02 X1.5 Y1. I0. J-.5 F30.
G00 X0. Y0.

Super basic, and it does work as expected, but it throws an Error 33. If I didn’t look at the terminal, I wouldn’t have even noticed it.

Curious if this program is still being actively developed. If so, I’d like to help. I’m no great Python coder, but I do have a CNC machine and I’m sort of a developer at work (iOS mainly, but others as the need arises…). I made a custom build of this last year with a modified UI that was better suited to touch screen operation. Let me know if there’s anything I can do to help!

  *   Mike

Harvie, 2019-12-01:
Error 33 is about problem in ARC geometry.

> Curious if this program is still being actively developed. If so, I’d like to help.

@michael-keymont-bose cool! we surely look for people to help improve the bCNC. there are some major issues, that nobody really knows how to tackle. Especially the toolpath rendering being slow as hell and viewport is not really 3D with smooth mouse rotation. I would like to see this fixed. But that's a rather big task. Anyway, i am willing to merge any pull request that would make sense to me and don't break anything important...